commaai/openpilot #982: Braking rate should be dependant on lead car

arne182, 2020-01-20:
In some situations where the lead car is creeping or if you are on a slope and the car is slowly accelerating, open pilot will brake to stop then notice that there is space to move again and start accelerating and then brake again. This is a quite rough behaviour and could be supplemented with radar data. We know the lead car's distance and speed and so we could use that to slowly or rapidly stop the car or accelerate the car.
Having a car speed away from you and you slowly deciding to move is not so nice. Having you look like you are bobbing your car around is bad, crashing into the lead car because your brake rate is fixed is just stupid.

pd0wm, 2020-01-20:
During normal driving MPC should take into account all of the information about lead distance, lead velocity, and will even do an estimation of their acceleration trajectory.

Is the behavior you are describing only present in stop and go? There are some state transitions there to go from a stopped state into normal driving. Maybe those transitions can be improved.

arne182, 2020-01-22:
Yes, mostly. There is one situation where the lead car is slowing down and then just at the end hits the brakes hard while self.long_control_state == LongCtrlState.stopping and the rate is fixed to 20% brake per second.

closed 2020-05-11 by pd0wm