commaai/openpilot #992: InitData.gitBranch has extra junk bytes in 0.7.1 release2

andyh2, 2020-01-22:
Reading initData.gitBranch from a 0.7.1 release2 log will throw `UnicodeDecodeError` as logged data in that field is `release2\xd0\x03\t\x99`. Let's figure out where the extra four bytes are coming from and fix the issue.

pd0wm, 2020-01-23:
Is this happening on all routes, or just a few?

andyh2, 2020-01-27:
All 0.7.1 routes running release2 are affected. Most forks seem to not have the issue. 

Also, the extra data at the end of the branch field is not consistently `\xd0\x03\t\x99`, I see many variants among 0.7.1 routes.

pd0wm, 2020-01-28:
Loggerd.cc makes the wrong assumption that the values from `read_db_value` are NULL terminated. Which there are not. This causes random junk to be appended.

pd0wm, 2020-01-28:
Should be fixed by: https://github.com/commaai/openpilot/commit/393c4987c8f1f8744cf2315f04ce9f14a5efc116

@andyh2, can you confirm?

andyh2, 2020-01-28:
LGTM and branch field is now clean. Verified in a log produced by running that commit in a local branch named 'release2'.

closed 2020-01-28 by andyh2