commaai/openpilot #996: Comma Two - 2020 Corolla - "Dashcam Mode Car Unrecognized"

noperdd, 2020-01-23:
**Describe the bug**
Installed Openpilot on a new Comma Two that arrived today.
Message onscreen is "Dashcam Mode Car Unrecognized"

**How to reproduce or log data**
Used a new install or openpilot. 
I had installed the dashcam version first, but then uninstalled it and followed instructions to get the latest from openpilot.comma.ai

**Expected behavior**
I think it should recognize the car.

**Device/Version information (please complete the following information):**
 - Device: [Comma Two]
 - Version: [0.7.1], 
 - Car make/model [Toyota Corolla 2020]

geohot, 2020-01-23:
Sadly, this is a known issue with certain trims of Corollas (LE and Hybrid I believe) due to limitations of fingerprinting 1.0. The good news, it's 100% a software issue, your comma two has all the hardware needed to fix this built in. You have two options:

1) Go on discord, discord.comma.ai and get help. There are forks which fix this right now, and people there should be able to help you get one of them.
2) Wait for 0.7.2 when we roll out fingerprinting 2.0 fixing this detection issue. @pd0wm We have a timeline on this?

Thanks for the report, we'll get this fixed as soon as possible.

thesoundofom, 2020-01-30:
If it's a Hybrid LE my fork will do the trick. In the process of rebasing it right now.

noperdd, 2020-02-08:
Thanks. I got help on discord from illumiN8i, helped get the fingerprint working. Looking forward to 7.2

noperdd, 2020-02-12:
This issue is fixed with the 7.2 release. Closing ticket. Thanks!

closed 2020-02-12 by noperdd