commaai/openpilot #1004: my open pill is asking me to connect to internet when it is already connected, I can used open pilot

camiloepp, 2020-01-25:


Debreo, 2020-01-26:
I had this issue too, but oddly enough after going to the Settings> network > WiFi settings and just reconnecting from more options and then hitting continue. I can’t explain it but it looks like it’s working now lol.

pd0wm, 2020-01-28:
What is your dongle id? I can check in the logs if the updater is running successfully.

You can also SSH in and run `scripts/update_now.sh`, and check for any output in the normal tmux window. You should see a reason why it's not running the updater.

pd0wm, 2020-02-13:
Closing this issue since no more information was provided. Feel free to open a new issue with more information.

closed 2020-02-13 by pd0wm

ch1derco, 2020-05-13:
Have the same problem. After about 2 weeks, the message "Immediately connect to the internet to check for updates. If you do not connect to the internet, open pilot won't engage in 0 days". Only a factory reset/new install of open pilot software solves the problem for another few weeks, then message comes again.

WidgeonRBLX, 2020-05-16:
This is also happening to me, tried everything in regard to rebooting, turning off WiFi/on again, will try reinstalling...

Zach-Attach, 2020-06-11:
I’m having the same issue as well with my EON, connected to a 2017 volt.