commaai/openpilot #1005: Turn on IR led for driver facing camera snapshot

HaraldSchafer, 2020-01-25:
When taking snapshots with the comma connect app, turn on the IR leds.

pd0wm, 2020-01-26:
@robbederks Does this require panda changes? We would need to be able to turn on IR leds when the panda is in power saving mode.

robbederks, 2020-01-26:
Yes, it is now not possible to turn on the IR in power savings. They use a lot of power (up to 6W), so we need to make sure that they definitely do not stay on.
I suggest putting them in the same logic as the safety mode (no heartbeat for a few seconds -> turn off)?

robbederks, 2020-01-28:
Needs changes in boardd too to make sure that they turn off when ignition goes low

pd0wm, 2020-01-30:
https://github.com/commaai/openpilot/commit/3a65206c8d0400fb22239c806429842236afac2b

pd0wm, 2020-01-30:
@ZwX1616 Do we need to do anything special on the camera side? Or will starting camerad always fire up the IR leds?

closed 2020-02-01 by pd0wm