commaai/openpilot #1010: Honda Civic Sedan 2019 not working on Comma2

ffhspa, 2020-01-26:
**Describe the bug**
On Comma2 OP doesnt work on a Honda Civic Sedan 2019 with Bosch Radar. 
ON disqord some Users (@Kev!n) tryed to Identify Cars Fingerprint. Then issue was, that following error appers on Comma2, "Gear not D"  
As this Car is Supported by OP, this seams to be a Bug. 

**How to reproduce or log data**
Car Fingerprint: 

228: 5, 229: 8, 513: 8, 576: 8, 577: 8, 579: 8, 580: 8, 582: 8, 583: 8, 585: 8, 586: 8, 640: 8, 641: 8, 642: 8, 643: 8, 644: 8, 645: 8, 646: 8, 647: 8, 648: 8, 649: 8, 650: 8, 651: 8, 652: 8, 653: 8, 654: 8, 655: 8, 656: 8, 657: 8, 658: 8, 659: 8, 660: 8, 661: 8, 662: 8, 663: 8, 712: 8, 713: 8, 714: 8, 715: 8, 716: 8, 717: 8, 718: 8, 719: 8, 720: 8, 721: 8, 722: 8, 723: 8, 724: 8, 725: 8, 726: 8, 727: 8, 728: 8, 729: 8, 730: 8, 731: 8, 732: 8, 733: 8, 734: 8, 735: 8, 736: 8, 737: 8, 738: 8, 739: 8, 740: 8, 741: 8, 742: 8, 743: 8, 744: 8, 745: 8, 746: 8, 747: 8, 748: 8, 749: 8, 750: 8, 751: 8, 752: 8, 753: 8, 754: 8, 755: 8, 756: 8, 757: 8, 758: 8, 759: 8, 760: 8, 761: 8, 762: 8, 763: 8, 764: 8, 765: 8, 766: 8, 767: 8, 829: 5, 1024: 4, 1040: 6, 1056: 8, 1088: 3, 1104: 8, 1641: 8, 1872: 8

Cabana Links: 

https://my.comma.ai/cabana/?route=d83f36766f8012a5%7C2020-01-25--09-42-43&exp=1611509140&sig=A%2F9iUBavlBPOw5M%2Both4Zl%2BjSNnxfnanvIuTgFvCc0I%3D&max=11&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd83f36766f8012a5%2Fc5e30ba0b150f3d9ba31b26e7efc729c_2020-01-25--09-42-43

https://my.comma.ai/cabana/?route=d83f36766f8012a5%7C2020-01-25--14-17-04&exp=1611509957&sig=AHqEqotiPOY1vnOQEtK9Ul8P9bnE4RFczv4FxeUiJT8%3D&max=2&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd83f36766f8012a5%2F6eca0fed46329049760d3a35ed72f58e_2020-01-25--14-17-04

https://my.comma.ai/cabana/?route=d83f36766f8012a5%7C2020-01-25--09-42-43&exp=1611509140&sig=A%2F9iUBavlBPOw5M%2Both4Zl%2BjSNnxfnanvIuTgFvCc0I%3D&max=11&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd83f36766f8012a5%2Fc5e30ba0b150f3d9ba31b26e7efc729c_2020-01-25--09-42-43

https://my.comma.ai/cabana/?route=d83f36766f8012a5%7C2020-01-25--14-17-04&exp=1611509957&sig=AHqEqotiPOY1vnOQEtK9Ul8P9bnE4RFczv4FxeUiJT8%3D&max=2&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd83f36766f8012a5%2F6eca0fed46329049760d3a35ed72f58e_2020-01-25--14-17-04

https://my.comma.ai/cabana/?route=d83f36766f8012a5%7C2020-01-25--13-56-30&exp=1611510188&sig=KCWvoygOJxZjGPlyJGyADvyiVRTdluk08hKsprs7deg%3D&max=4&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd83f36766f8012a5%2F595fbce2715f44015a39a298d80ee493_2020-01-25--13-56-30

**Expected behavior**
OP Should Engage on Honda Civic Sedan 2019 

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Version: 0.7.1 release2
 - Car: Honda Civic Sedan 2019

geohot, 2020-01-27:
We will address this in 0.7.2

ffhspa, 2020-01-27:
if you need any further Information, my disqord username is: hondacivicwyde

Thanks a lot!

ffhspa, 2020-01-31:
My Car has been Portet susccefully to OP thanks to Chris!
He will put in the next time a Pull Request for the changes he made. 

Thanks!

pd0wm, 2020-02-02:
What fork are you currently running? I'd like to check the modifications made to it.

I also made this PR with your FW versions, so maybe that's all that's needed to support your car: https://github.com/commaai/openpilot/pull/1033

ffhspa, 2020-02-03:
Chris portet my Car succefully to OP. 
Here is the Link to the Fork he made: https://github.com/csouers/openpilot/tree/19-civic-1.6

pd0wm, 2020-02-03:
Perfect!

closed 2020-02-03 by pd0wm