commaai/openpilot #1012: Toyota Sienna 2020 fingerprint

ScottMcJ, 2020-01-27:
If useful for someone keeping the fingerprint data for vehicles, I have the fingerprint below for my 2020 Toyota Sienna AWD Limited Premium, which is working fine for lane assist and cruise control to zero (albeit trying to figure out why I am getting some erratic "cruise error, restart car" (for which I have to turn off car, plug in the DSU, turn on the car turn off the car, unplug the DSU and then I am back in business), and a weirdly loud strumming/stuttering vibration from the front passenger side as it engages braking down to, and at, 0 mph).  

I slapped the fingerprint on on top of the 2018 Sienna fingerprint, and that worked.  But it would be lovely to have it incorporated into the core code.

My fingerprint is:

number of messages 147:
fingerprint 36: 8, 37: 8, 114: 5, 119: 6, 120: 4, 170: 8, 180: 8, 186: 4, 238: 4, 426: 6, 452: 8, 464: 8, 466: 8, 467: 8, 544: 4, 545: 5, 548: 8, 550: 8, 552: 4, 562: 4, 608: 8, 610: 5, 643: 7, 705: 8, 725: 2, 740: 5, 764: 8, 800: 8, 818: 8, 822: 8, 824: 8, 835: 8, 836: 8, 849: 4, 869: 7, 870: 7, 871: 2, 888: 8, 889: 8, 891: 8, 896: 8, 900: 6, 902: 6, 905: 8, 911: 8, 916: 1, 918: 7, 921: 8, 933: 8, 944: 6, 945: 8, 951: 8, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1008: 2, 1014: 8, 1017: 8, 1041: 8, 1042: 8, 1043: 8, 1056: 8, 1059: 1, 1076: 8, 1077: 8, 1114: 8, 1160: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1176: 8, 1177: 8, 1178: 8, 1179: 8, 1180: 8, 1181: 8, 1182: 8, 1183: 8, 1191: 8, 1192: 8, 1196: 8, 1197: 8, 1198: 8, 1199: 8, 1200: 8, 1201: 8, 1202: 8, 1203: 8, 1212: 8, 1227: 8, 1228: 8, 1235: 8, 1237: 8, 1263: 8, 1264: 8, 1279: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1561: 8, 1562: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1584: 8, 1589: 8, 1592: 8, 1593: 8, 1595: 8, 1656: 8, 1664: 8, 1666: 8, 1667: 8, 1728: 8, 1745: 8, 1779: 8, 1792: 8, 1800: 8, 1872: 8, 1880: 8, 1904: 8, 1912: 8, 1937: 8, 1945: 8, 1953: 8, 1961: 8, 1968: 8, 1990: 8, 1998: 8, 2015: 8, 2016: 8, 2024: 8 

Cheers!
-Scott

pd0wm, 2020-02-02:
Can you submit this as PR, together with a cabana drive with the stock system enabled? See https://github.com/commaai/openpilot/pull/1032 as an example.

ScottMcJ, 2020-02-04:
Here is a drive Link, if useful?
https://my.comma.ai/2dc5556833c421c2/1580774974087/1580776463514

pd0wm, 2020-02-06:
Please submit a pull request adding this fingerprint!

ScottMcJ, 2020-02-06:
I have not made my own branch; as noted above, I just brute-forced it in on my local copy over the other Sienna fingerprint.  I do not know how to submit a pull request when this is a feature request that I am suggesting someone else might like to bring into the code base to widen compatibility.

ScottMcJ, 2020-02-14:
0.7.2 needed me to re-fingerprint.  If anyone wanted to run with this add, here is my new fingerprint generated and a new drive:

36: 8, 37: 8, 114: 5, 119: 6, 120: 4, 170: 8, 180: 8, 186: 4, 238: 4, 426: 6, 452: 8, 464: 8, 466: 8, 467: 8, 544: 4, 545: 5, 548: 8, 550: 8, 552: 4, 562: 4, 608: 8, 610: 5, 643: 7, 705: 8, 725: 2, 740: 5, 764: 8, 800: 8, 818: 8, 822: 8, 824: 8, 835: 8, 836: 8, 849: 4, 869: 7, 870: 7, 871: 2, 888: 8, 889: 8, 891: 8, 896: 8, 900: 6, 902: 6, 905: 8, 911: 8, 916: 1, 918: 7, 921: 8, 933: 8, 944: 6, 945: 8, 951: 8, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1008: 2, 1014: 8, 1017: 8, 1041: 8, 1042: 8, 1043: 8, 1056: 8, 1059: 1, 1076: 8, 1077: 8, 1114: 8, 1160: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1176: 8, 1177: 8, 1178: 8, 1179: 8, 1180: 8, 1181: 8, 1182: 8, 1183: 8, 1191: 8, 1192: 8, 1196: 8, 1197: 8, 1198: 8, 1199: 8, 1200: 8, 1201: 8, 1202: 8, 1203: 8, 1212: 8, 1227: 8, 1228: 8, 1235: 8, 1237: 8, 1263: 8, 1264: 8, 1279: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1561: 8, 1562: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1584: 8, 1589: 8, 1592: 8, 1593: 8, 1595: 8, 1656: 8, 1664: 8, 1666: 8, 1667: 8, 1728: 8, 1745: 8, 1779: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8

https://my.comma.ai/cabana/?route=2dc5556833c421c2%7C2020-02-14--07-46-21&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F2dc5556833c421c2%2Ff1811e611f28780191b3046e073d9e63_2020-02-14--07-46-21&seekTime=2

pd0wm, 2020-02-14:
We're no longer adding fingerprints for Toyota. However, I added the FW versions from your logs to openpilot: https://github.com/commaai/openpilot/commit/cd2c1d79375eb01c621decaa7179ffd590a9e407

Your car should now be recognized correctly.

closed 2020-02-14 by pd0wm

familycio, 2020-05-21:
@ScottMcJ Do you still have the issue of  "cruise error, restart car" and vibration when brake ? I'm thinking to buy a 2020 Sienna - appreciate your info!

ScottMcJ, 2020-05-21:
Good morning!  The generous pd0wm added my engine code (?) to the
code-base, so I would not have to re-fingerprint every update, but I am not
sure whether that means it covers all the siennas or just mine?  We would
have to ask.  That took care of the cruise error problems.  I've been good
through three software updates.

The noise when braking to a stop is there.  The braking itself is fine and
smooth.  After unplugging the DSS, it brakes down to zero and resumes from
zero (which was a major irritant when I got the sienna; what kind of
ridiculous adaptive cruise control cuts at at 27mph? useless for
traffic!).  I do not know if it is the ABS actuator or what, but some
mechanical doohickey on the passenger side of the engine makes a thrumming
noise at about 1mph and again at full stop.  It's weird, but not too
intrusive and I am living with it.  You wouldn't hear it unless the radio
was off.

-S.