commaai/openpilot #1018: Fingerprinting for Toyota 2019 Corolla Hybrid - Hatchback Australia

adam-nelson, 2020-01-28:
Hi

Just want to contribute my fingerprinting for Toyota 2019 Corolla Hybrid - Hatchback Australia delivered. SNip3D on Discord can post his sedan fingerprint

fingerprint 36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 401: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4,
560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 742: 8, 743: 8, 761: 8, 764: 8, 765: 8, 800: 8, 810: 2, 812: 8, 824: 8, 829: 2, 830: 7, 835: 8, 836: 8, 865: 8, 869: 7, 870: 7, 871: 2, 877: 8, 881: 8, 896: 8, 898: 8, 900: 6, 902: 6, 905: 8, 921: 8, 933: 8, 934: 8, 935: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059:
1, 1071: 8, 1076: 8, 1077: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1172: 8, 1235: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1595: 8, 1745: 8, 1775: 8, 1779: 8, 1786: 8, 1787: 8, 1788: 8, 1789: 8, 1792: 8, 1800: 8, 1872: 8, 1880: 8, 1904: 8, 1912: 8, 1937: 8, 1945: 8, 1953: 8, 1961: 8, 1968: 8, 1976: 8, 1990: 8, 1998: 8, 2015: 8, 2016: 8, 2024: 8

pd0wm, 2020-02-02:
Can you submit this as PR, together with a cabana drive with the stock system enabled? See https://github.com/commaai/openpilot/pull/1032 as an example.

adam-nelson, 2020-02-02:
I think my biggest complaint about fingerprinting is there no way of nesting fingerprints under a certain car type

pd0wm, 2020-02-02:
What do you mean by nesting? You can add multiple fingerprints for the same car model.

adam-nelson, 2020-02-07:
Nesting or making a car of child of a parent car

My fingerprint doesnt match a Corolla TSS 2.0 (parent) and need a fingerprint of the Australian Model (child)

pd0wm, 2020-02-10:
Each car already has a list for the fingerprints. You can just add it to the list for that specific model.

pd0wm, 2020-02-11:
Just pushed your firmware versions to master and release2: https://github.com/commaai/openpilot/commit/aea4bee60eb6c7e7f43db31df6b95e136f8d48e2

I assume your fingerprint issues are now solved.

closed 2020-02-11 by pd0wm

cameron20020, 2020-04-15:
Had to manually add my Australian Corolla Hybrid fingerprint to my comma 2 running default openpilot today.

Done like:
https://github.com/cameron20020/openpilot/blob/c39f517042f50d2eaa180a3321602519a31aa4ea/selfdrive/car/toyota/values.py#L220

Works, but I couldn't figure out how to do a pull request. Just want this fixed so no one else has this problem.

adam-nelson, 2020-04-15:
With the latest fingerprinting you shouldn't have to make changes to
values.py

cameron20020, 2020-04-15:
I did the normal install from "https://openpilot.comma.ai" during setup.
But it was showing vehicle unrecognisable until I edited the values.py

Am I doing something wrong? or is there a newer/other version I should grab?

adam-nelson, 2020-04-15:
Did you get the 2020 corolla hybrid hatchback?

cameron20020, 2020-04-15:
Correct.
Mine is a ZWE211R, the new one built on TNGA.

adam-nelson, 2020-04-15:
Ahh I see mines a 2019 version

Should be identical. Are you in Australia?

cameron20020, 2020-04-16:
Issue is resolved, but isn't in release 2 yet, but will be.

See:
https://discordapp.com/channels/469524606043160576/524592892627517450/700151077957533736