commaai/openpilot #1042: Tests are too slow

geohot, 2020-02-02:
Why does build take 13 minutes? Can we cache better?

pd0wm, 2020-02-02:
We can manually push a docker container to the github package registry that can be used a base for the build. We have to manually update that when we make big changes to the dockerfile.

Unfortunately Github actions or Azure pipelines doesn't let you persist the docker cache between workflows.

geohot, 2020-02-02:
That sounds reasonable, as long as we have an easy script for it. I don't think we change those files much.

quillford, 2020-02-02:
I think CircleCI does free docker layer caching for open source projects.

pd0wm, 2020-02-03:
Pushed a container to dockerhub.

I also tried using the github docker registry, but you need to login before you can pull any images, even for public repos. Logging in requires a token which doesn't work on forks.

pd0wm, 2020-02-03:
Added caching. Went from about 14 minutes to 10 minutes.

For some reason the caching breaks when it loads the Pipfile, it doesn't see that that the file is the same.

pd0wm, 2020-02-06:
That was built on my local machine. It's not exactly clear from the documentation what metadata it uses.

quillford, 2020-02-10:
I was able to shave off [~3 minutes](https://github.com/quillford/openpilot/runs/437115077?check_suite_focus=true) (9.5 to 6) after using the image built in CI as the cache source. Not sure what metadata they're using, but I think going forward pushing images built in CI is the way to go, and they're already uploaded as artifacts.

pd0wm, 2020-02-11:
Thanks! Think there is not much more to gain now in the build step.

closed 2020-02-11 by pd0wm