commaai/openpilot #1051: Make HEVC files correct framerate

HaraldSchafer, 2020-02-04:
Saved road facing camera hevc files are saved at 50FPS, but should be 20FPS

HaraldSchafer, 2020-02-04:
probably somewhere in here https://github.com/commaai/openpilot/blob/master/selfdrive/loggerd/encoder.c

nelsonjchen, 2020-02-04:
Just linking this here in-case somebody from the future is trying to see what's up with VLC and wondering why repro isn't happening. 

https://trac.videolan.org/vlc/ticket/22681#comment:1

GitHub View of commits mentioned:

https://github.com/videolan/vlc/commit/fe88719c808cda0e91fd4d3da5c67218cb3e73f5

https://github.com/videolan/vlc/commit/f851387a45c3147e13437765206f0dca179f2c4a

geohot, 2020-02-04:
Err, there isn't really a way to do this. There's not a wrong frame rate, there's no frame rate, so it becomes the player default. The hevc files are not in a container, adding one would change the format, so will not fix.

closed 2020-02-04 by geohot