commaai/openpilot #1070: Multilanguage support

geohot, 2020-02-09:
Before openpilot is 1.0 software, it needs support for multiple languages. We'd need a nice generic way to do this.

xx979xx, 2020-02-19:
I started working on this (not ready yet) https://github.com/xx979xx/openpilot/blob/i18n/selfdrive/controls/lib/alerts.py
plz everyone feel free to critique and contribute

fusk-l, 2020-02-25:
For the language part, are you going to use something like transifex or similar? I would be more than happy to help with translation.

s8321414, 2020-08-07:
When i18n and l10n support is ready, I can help to translate it to zh_TW.

caginbektas, 2020-08-10:
I would love to start contributing by this. Count me in!

geohot, 2020-08-10:
The UI rewrite should be done with this in mind.

xx979xx, 2020-10-28:
@fusk-l @s8321414 @caginbektas here all Openpilot translatable texts up-to-date
[messages.po](https://github.com/xx979xx/openpilot/blob/i18n/selfdrive/assets/locales/messages.po)

fusk-l, 2020-10-28:
> 
> 
> @fusk-l @s8321414 @caginbektas here all Openpilot translatable texts up-to-date
> [messages.po](https://github.com/xx979xx/openpilot/blob/i18n/selfdrive/assets/locales/messages.po)
> here is the code(not ready):
> [i18n.py](https://github.com/xx979xx/openpilot/blob/i18n/selfdrive/common/i18n.py)
> [i18n.h](https://github.com/xx979xx/openpilot/blob/i18n/selfdrive/common/i18n.h)

Just to be sure, we just copy everything, translate msgid and reupload where?

eFiniLan, 2020-10-28:
@xx979xx does your ui.cc/sidebar.cc translation working? I tried a while ago and couldnt get it to work :(

xx979xx, 2020-10-30:
@eFiniLan I have not tried it yet
needless to say, this is my first ever C++ code, I am doing it for fun (and learning C++ which I hate so much)
soon you will find out

fusk-l, 2020-10-30:
Whenever there's an explanation of how this works, i'll be ready. Have already looked through some of the strings.

xx979xx, 2020-10-30:
@eFiniLan you are right, ui does not translate. I think we have to install gettext package from [termux-packages repo](https://github.com/termux/termux-packages/tree/master/packages/gettext)
I tried but I got this error:
```
$ pkg install gettext
This must NOT be run as root as it will break your environment (root will be required for nearly everything due to changed file permissions)
```

grafst, 2021-08-30:
I'm willing to translate into german.

KristofKekesi, 2021-08-31:
I can translate it to Hungarian

kengggg, 2021-09-01:
I'm happy to translate into Thai. 👌

a-maccormack, 2022-02-24:
I can translate to Spanish

ponzu07, 2022-02-24:
I can translate Japanese.

royjr, 2022-03-08:
I got Arabic!

sshane, 2022-07-09:
Merged base multilanguage support into master for offroad settings and onroad static text elements.

- [ ] We would still like to dynamically translate without restarting UI
- [ ] Translate alerts (dynamic text requires injecting our own translations with custom contexts into the TS files)

sshane, 2022-07-09:
Moved improvements to https://github.com/commaai/openpilot/issues/25092

closed 2022-07-09 by sshane