commaai/openpilot #1081: Chrysler Pacifica PHEV 2019 "Dashcam Mode - Car Unrecognized"

igor-fedulov, 2020-02-12:
**Describe the bug**
After upgrade to openpilot 0.7.2 the autopilot functionality disappeared, the EON only in dashcam mode.

**How to reproduce or log data**
Here is the fingerprint from the console:

fingerprint 168: 8, 257: 5, 258: 8, 264: 8, 268: 8, 270: 8, 274: 2, 280: 8, 284: 8, 288: 7, 290: 6, 291: 8, 292: 8, 294: 8, 300: 8, 308: 8, 320: 8, 324: 8, 331: 8, 332: 8, 344: 8, 368: 8, 376: 3, 384: 8, 388: 4, 448: 6, 456: 4, 464: 8, 469: 8, 480: 8, 500: 8, 501: 8, 512: 8, 514: 8, 515: 7, 516: 7, 517: 7, 518: 7, 520: 8, 528: 8, 532: 8, 544: 8, 557: 8, 559: 8, 560: 8, 564: 8, 571: 3, 579: 8, 584: 8, 608: 8, 624: 8, 625: 8, 632: 8, 639: 8, 653: 8, 654: 8, 655: 8, 660: 8, 669: 3, 671: 8, 672: 8, 680: 8, 701: 8, 703: 8, 704: 8, 705: 8, 706: 8, 709: 8, 710: 8, 719: 8, 720: 6, 736: 8, 737: 8, 746: 5, 752: 2, 754: 8, 760: 8, 764: 8, 766: 8, 770: 8, 773: 8, 779: 8, 782: 8, 784: 8, 792: 8, 799: 8, 800: 8, 804: 8, 808: 8, 816: 8, 817: 8, 820: 8, 825: 2, 826: 8, 832: 8, 838: 2, 848: 8, 853: 8, 856: 4, 860: 6, 863: 8, 878: 8, 882: 8, 897: 8, 906: 8, 908: 8, 924: 8, 926: 3, 929: 8, 937: 8, 938: 8, 939: 8, 940: 8, 941: 8, 942: 8, 943: 8, 947: 8, 948: 8, 958: 8, 959: 8, 962: 8, 969: 4, 973: 8, 974: 5, 979: 8, 980: 8, 981: 8, 982: 8, 983: 8, 984: 8, 992: 8, 993: 7, 995: 8, 996: 8, 1000: 8, 1001: 8, 1002: 8, 1003: 8, 1008: 8, 1009: 8, 1010: 8, 1011: 8, 1012: 8, 1013: 8, 1014: 8, 1015: 8, 1024: 8, 1025: 8, 1026: 8, 1031: 8, 1033: 8, 1050: 8, 1059: 8, 1082: 8, 1083: 8, 1098: 8, 1100: 8
number of messages 157:

**Expected behavior**
The autopilot used to work before just fine

**Device/Version information (please complete the following information):**
 - Device: EON
 - Version: 0.7.2
 - Car make/model Chrysler Pacifica PHEV 2019

**Additional context**
Not applicable

pd0wm, 2020-02-13:
Can you share your dongle id or a route where this happens?

arne182, 2020-02-13:
>>> import fingerprint_helper                                >>> fingerprint_helper.find_matches()                        Enter fingerprint: {168: 8, 257: 5, 258: 8, 264: 8, 268: 8, 270: 8, 274: 2, 280: 8, 284: 8, 288: 7, 290: 6, 291: 8, 292: 8, 294: 8, 300: 8, 308: 8, 320: 8, 324: 8, 331: 8, 332: 8, 344: 8, 368: 8, 376: 3, 384: 8, 388: 4, 448: 6, 456: 4, 464: 8, 469: 8, 480: 8, 500: 8, 501: 8, 512: 8, 514: 8, 515: 7, 516: 7, 517: 7, 518: 7, 520: 8, 528: 8, 532: 8, 544: 8, 557: 8, 559: 8, 560: 8, 564: 8, 571: 3, 579: 8, 584: 8, 608: 8, 624: 8, 625: 8, 632: 8, 639: 8, 653: 8, 654: 8, 655: 8, 660: 8, 669: 3, 671: 8, 672: 8, 680: 8, 701: 8, 703: 8, 704: 8, 705: 8, 706: 8, 709: 8, 710: 8, 719: 8, 720: 6, 736: 8, 737: 8, 746: 5, 752: 2, 754: 8, 760: 8, 764: 8, 766: 8, 770: 8, 773: 8, 779: 8, 782: 8, 784: 8, 792: 8, 799: 8, 800: 8, 804: 8, 808: 8, 816: 8, 817: 8, 820: 8, 825: 2, 826: 8, 832: 8, 838: 2, 848: 8, 853: 8, 856: 4, 860: 6, 863: 8, 878: 8, 882: 8, 897: 8, 906: 8, 908: 8, 924: 8, 926: 3, 929: 8, 937: 8, 938: 8, 939: 8, 940: 8, 941: 8, 942: 8, 943: 8, 947: 8, 948: 8, 958: 8, 959: 8, 962: 8, 969: 4, 973: 8, 974: 5, 979: 8, 980: 8, 981: 8, 982: 8, 983: 8, 984: 8, 992: 8, 993: 7, 995: 8, 996: 8, 1000: 8, 1001: 8, 1002: 8, 1003: 8, 1008: 8, 1009: 8, 1010: 8, 1011: 8, 1012: 8, 1013: 8, 1014: 8, 1015: 8, 1024: 8, 1025: 8, 1026: 8, 1031: 8, 1033: 8, 1050: 8, 1059: 8, 1082: 8, 1083: 8, 1098: 8, 1100: 8}
Exact fingerprint match!
CHRYSLER PACIFICA HYBRID 2019
15 less key/value pairs

Looks to be perfectly working on my fork

igor-fedulov, 2020-02-19:
> Looks to be perfectly working on my fork

Is there a fork I can switch to, or is this fix will be merged into release2 soon? Please advise!

arne182, 2020-02-19:
Those fingerprints come from my fork. Having collected and merged fingerprints over many months and many users. You can try it out and PR my merged fingerprint. Maybe it will be considered

igor-fedulov, 2020-02-19:
> Those fingerprints come from my fork. Having collected and merged fingerprints over many months and many users. You can try it out and PR my merged fingerprint. Maybe it will be considered

Do I get 072-clean or release4?

arne182, 2020-02-19:
Go release4 go!

igor-fedulov, 2020-02-19:
> Go release4 go!

Will try tonight and report back. Thanks!

igor-fedulov, 2020-02-19:
Tried just now including flashing panda. Still the same message - "Dashcam Mode - Car Unrecognized". Have your picture now in the bottom left corner though!

arne182, 2020-02-19:
Sadly you tried it without internet access and I was not able to get the crash log of why it went to unrecognized car. Could you try it again with a hotspot or wifi?

arne182, 2020-02-19:
I see what is happening, I updated the helper with the 2020 hybrid and it looks like
Exact fingerprint match!
CHRYSLER PACIFICA HYBRID 2019
15 less key/value pairs

Exact fingerprint match!
CHRYSLER PACIFICA HYBRID 2020
13 less key/value pairs
Your car is a subset of the two fingerprints

arne182, 2020-02-19:
I see no need for them to be split. There are no conflicting messages.

pd0wm, 2020-02-20:
Please leave the discussion of forks to #custom-forks on discord.

@igor-fedulov Can you share your dongle id? Then I can add the fingerprint, and run a replay of on of your drives to verify that it actually works.

@NeilEugene Thanks for sharing!

pd0wm, 2020-02-20:
As suggested by @arne182 the 2019 and 2020 are the same. I made this pr to merge them https://github.com/commaai/openpilot/pull/1142

Let me know it running that PR solves your issues!

igor-fedulov, 2020-02-21:
Good evening. I reinstalled the OpenPilot from workbench and reflashed Panda. This morning I got LKAS fault and on subsequent drive the problem went away. The dongle id is f6a0b679cf4fa9d2

pd0wm, 2020-02-21:
Does running the `master-ci` branch fix your fingerprinting problems?

igor-fedulov, 2020-02-28:
Another observation. The EON upgraded to 0.7.3 automatically. The autopilot was working fine for two days or so and today started again giving unrecognized car dashcam mode message on start. I did the reinstall open pilot through the console - didn’t help. I then went to do a full factory reset holding volume down and power and reinstalled open pilot and the problem went away. I’m back on 0.7.3 and it again works.

NeilEugene, 2020-02-28:
Mine also still says unrecognized car after upgrade. If next upgrade does
not fix dashcam mode I'll have to try factory reset as well.

Tks for the tip
Neil

igor-fedulov, 2020-02-28:
This morning the EON is back to Dashcam mode. So hard reset didn’t help. Sounds like car signature has variations.

Jejunum, 2020-02-29:
Total noob, just got a comma 2 for a 2019 Pacifica hybrid. Same error message after installing openPilot... Can anyone help me as well :)

pd0wm, 2020-03-03:
The fix will go out in 0.7.4, so until now you have to run the `master-ci` if you want the fix.

closed 2020-03-03 by pd0wm

Jejunum, 2020-03-03:
Can you explain what that means? And also howdo I understand the difference
between versions?

igor-fedulov, 2020-03-05:
@pd0wn Verified fixed using "master-ci" branch. Thanks!

igor-fedulov, 2020-03-05:
@Jejunum you can read on how to install fork or another branch here: https://medium.com/@jfrux/comma-eon-installing-a-fork-of-openpilot-5c2b5c134b4b