commaai/openpilot #1085: ACC can be bypassed on Toyota by holding down cruise on/off button

illumiN8i, 2020-02-12:
**Describe the bug**
ACC can by bypassed on Toyota. Openpilot allows engagement when ACC is disabled. This bug was already fixed on Honda https://github.com/commaai/openpilot/pull/770 

**How to reproduce or log data**
Press and hold down the cruise on/off button. ACC will be disabled and the car will switch to non-adaptive cruise control. Press set to engage. Openpilot will control steering, but ACC will not function.

**Expected behavior**
Openpilot should not be allowed to be engaged without ACC.

**Device/Version information:**
 - Device: EON Gold
 - Version: 0.7.2
 - Car make/model: 2018 Toyota Prius Prime

**Additional context**
This loophole has been tested on the three major types of Toyota: DSU connected and disconnected, Nodsu, and TSS 2.0

I suspect that this signal could be used to detect when non-adaptive cruise control is enabled:
[image]

I believe CRUISE_CONTROL_STATE 1.0 is non-adaptive cruise control. This could be used to detect that state.

I have demonstrated using the loophole in this drive https://my.comma.ai/cabana/?route=b29e3bc918751697%7C2020-02-12--12-04-27&exp=1613076855&sig=Z3uBeA6LE4ovlWgKaSEomaR6j8KdcUViZ8176dMW6Jk%3D&max=13&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fb29e3bc918751697%2F1fadd736273c29c679b40bab0ff87058_2020-02-12--12-04-27

pd0wm, 2020-02-13:
I'm also seeing states 5 and 6. Can you try to engage in both modes and see what the state goes to?

This is what the DBC file says for that field:
`VAL_ 921 CRUISE_CONTROL_STATE 2 "disabled" 11 "hold" 10 "hold_waiting_user_cmd" 6 "enabled" 5 "faulted";`

Looks like 5 doesn't mean faulted since I see it during normal operation.

arne182, 2020-02-13:
Do you mean hold the canceI button and then press the set button again? Sounds quite difficult with the stock stalk.
I think I have been able to get into this mode by holding the brake down for long so that the electronic handbrake and then engage. Or sometimes when directly behind a lead and speed very close to 0 then the brakes lock and do not release. Or my waiting mode where set speed up does not work until you apply gas to get it out of the waiting mode. But maybe those are maybe different variations.

pd0wm, 2020-02-14:
I assume this is with TSS2 where there is a cruise on/off button on the steering wheel.

Hubblesphere, 2020-02-14:
Yes the TSS2.0 operation is holding the Cruise_ON button for ~1 second and it initially goes to state 2 then drops to 1. Another press resets it to 0 and you have to hold again. Currently OpenPilot does not car which cruise state it's in as long as cruise is on.

pd0wm, 2020-02-14:
I'm saying that when I looked at out Corolla data I saw other states when system was actually engaged. So we can't check for state == 2. Therefore we need to reverse engineer more of the states before we can implement a fix.

Hubblesphere, 2020-02-14:
It has many states for the difference between cruise modes as well as if there is a lead vehicle or not and separate engage/disengaged states for both ACC and constant speed cruise. Going to verify more today but from what we have seen so far it will probably be best to check that it is !=1 or 3 which are the two states of operation for constant speed (we are just calling bad cruise to simplify). I'll do more testing to verify those are the only states it operates in. Then we can disqualify based on those states only.

cydia2020, 2020-02-17:
Currently if you engage openpilot in this mode (holding down stalk button) and press the acceleration pedal, openpilot will disengage and the car would just cruise without radar

To some extent one can argue these two cruise modes are quite useful when you don't want to be held back by traffic that is only a bit slower than you but still want openpilot to steer for you.

Hubblesphere, 2020-02-17:
Can confirm that while using constant speed cruise OP will disengage on gas but constant speed cruise stays active. I would call it a safety issue rather than a useful feature. Someone could easily get confused about what is engaged and what is disengaged and also be unsure if radar cruise is active or not.

cydia2020, 2020-02-17:
> Someone could easily get confused about what is engaged and what is disengaged and also be unsure if radar cruise is active or not.

You can easily distinguish different modes from the information displayed either on the dash or the hud, with RADAR cruise on, the dash/hud will display three bars (and a vehicle if a lead is present); with RADAR cruise off and openpilot engaged, it will just display a speed, and openpilot's frame is green; if only cruise control is engaged, the hud will display your set speed and openpilot's frame is dark blue.

pd0wm, 2020-02-18:
Openpilot will never support different driving modes. It's on or off.

@Hubblesphere let us know if you find out more about the states.

Hubblesphere, 2020-02-18:
So far with TSS2.0 I've documented these states:

VAL_ 921 CRUISE_CONTROL_STATE 2 0 "off" 1 "constant_speed_disabled" 2 "acc_disabled" 3 "constant_speed_enabled"  5 "acc_enabled_no_lead" 6 "acc_enabled_with_lead" 10 "hold_waiting_user_cmd" 11 "hold";

Both with Stock enabled and OpenPilot enabled the constant speed cruise is controlled between states 1 and 3 when cruise is on. If that holds true for all other Toyotas might be easy to apply across the entire make.

adeebshihadeh, 2020-06-16:
#1708 adds the support code necessary for this. I'd prefer to use the `PCM_CRUISE` message since all supported toyotas use it. I started with one state in that PR, but there might be more (it's a 4 bit signal).

PR #2708 (2021-01-04): adaptive cruise can no longer be bypassed on toyota
**Description**
This PR aims to fix #1085
**Verification**
Went on a drive, held down the cruise button to enter non-adaptive cruise mode on my Prius 2018. openpilot as expected will not enable after this fix is applied.
**Route**
Route: 399b1cca765e80b4|2020-12-07--01-11-46
closed 2021-01-04 by pd0wm