commaai/openpilot #1135: Hyundai: simplify gasPressed logic

rbiasini, 2020-02-20:
Currently, the logic is:
```
if (cp.vl["TCS13"]["DriverOverride"] == 0 and cp.vl["TCS13"]['ACC_REQ'] == 1):
  pedal_gas = 0
else:
  pedal_gas = cp.vl["EMS12"]['TPS']
ret.gasPressed = pedal_gas > 1e-3
```
This depends on too many signals. Moreover, I believe gasPressed only becomes true when the driver acceleration request exceeds the one from ACC.

Instead, it would be better to use a simple "gas pedal position" signal.
@xx979xx, any recommendation?

TK211X, 2020-02-20:
Such a signal has only been discovered on EVs

xx979xx, 2020-02-21:
@rbiasini in theory, it should be like this:
```
ret.gas = cp.vl["EMS12"]['PV_AV_CAN']
ret.gasPressed = bool(ret.gas)
```
need test across all working cars

TK211X, 2020-02-21:
[image]

rbiasini, 2020-02-21:
@TK211X , based on what you posted, seems like PV_AV_CAN is "user+computer", no?

xx979xx, 2020-02-21:
I see, yes

TK211X, 2020-02-21:
I’ll establish a demonstration.

xx979xx, 2020-02-21:
PV_AV_CAN vs TPS
[image]

TK211X, 2020-02-21:
https://my.comma.ai/cabana/?route=53ac3251e03f95d7%7C2020-02-21--15-40-52&exp=1613853905&sig=eGuSmOTiRhRemSqQsobEFZViLMBdV%2BrGdhIW8I4XfIc%3D&max=1&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F53ac3251e03f95d7%2Fa2c993d96520119ee34ca5dcfedfaac7_2020-02-21--15-40-52

Near the start ACC is enabled and set.

Past the 55MPH speed limit sign going onto the bridge I floor the accelerator pedal.

At the end of the bridge I hit the brakes.

TK211X, 2020-02-22:
[image]
Green = SCC Set
Yellow = Driver Accelerator Pedal Max Action
Red = Driver Brake Action

TK211X, 2020-02-22:
Actually....... I've spent some time comparing values from the various EMS messages and found a signal that may be of interest to you @rbiasini 

`CF_Ems_AclAct` (EMS16)

It appears to be a flag that's specific to driver accelerator pedal actuation.

[image]
[image]

TK211X, 2020-02-22:
`CF_Ems_AclAct` (EMS16)

Literally appears to be a signal that says "hey, the driver is using the gas pedal."

Can't get more simple than that.

One thing to keep in mind is that not all vehicles have the same power-plant signals as some messages accomplish specific missions for that model.

Given the nature of this discovery and the fact that HKMC tends to keep signals and their formats universal, albeit, sometimes grouped differently under a different CAN message. I'm sure that finding a similar message for a given model is possible if EMS16 is not present for said model. For example as stated previously in EV's there exists an actual driver accelerator position sensor etc.

xx979xx, 2020-02-22:
what a wonderful job @TK211X 
so here you have it @rbiasini 
```
ret.gas = cp.vl["EMS12"]['PV_AV_CAN'] / 100
ret.gasPressed = cp.vl["EMS16"]["CF_Ems_AclAct"]
```

rbiasini, 2020-02-22:
this is great, @TK211X !
Yeah, I'm going to change both gas and gasPressed as you wrote, @xx979xx . 
Nice.

closed 2020-02-22 by rbiasini

rbiasini, 2020-02-23:
merged changes. If you end up driving the code, let me know if you have any issues. Thanks!
https://github.com/commaai/openpilot/pull/1152

TK211X, 2020-02-23:
lol, when is the harness being released. Doubt there would be a reason to use official until it comes out. I hear KU7 said official didn’t even work when he tested it.

TK211X, 2020-02-23:
[image]

rbiasini, 2020-02-23:
@TK211X , it's ok if you guys are on a custom fork. But I recommend to keep panda code in synch (which is affected by this change).