commaai/openpilot #1156: “Openpilot not available, issue with communication processes”

djjohnny1093, 2020-02-23:
Got my comma 2 in mail, installed everything correctly, rechecked all connections. Open pilot wont engage when pressing set. I get the above titled error. 

Tried from rolling start.
Tried on highway at various speeds.

2017 Honda CRV Touring.

Restarted device and car several times...

Panda box says “Good Gps” 

Pushed obd connector in till i heard a click. 

Calibration above 16mph worked.

No pins are bent on anything. Brand new straight out the box.

A little disappointed. 

Please somebody tell me i didn't buy a $1300 paperweight.

arne182, 2020-02-23:
Don't stress too much. Reach out to us on discord and we will gladly help get you up and running.

pd0wm, 2020-02-27:
Debugged this issue through discord. Seems like the driver monitoring model was not running. I suspect an issue with the DSP on your comma 2. We will replace the hardware.

closed 2020-02-27 by pd0wm