commaai/openpilot #1162: Fix complete system lags where every service hangs -- $1000 bounty

pd0wm, 2020-02-24:
Very rarely we have seen complete system lockup of about 0.2 to 1 second. See for example the time between packets in this plot:
[image]

Usually this causes a CAN error out of nowhere and a disengagement. One user reported that it always happens at the same point in his commute when he loses cellular connection.

My theory is that it is related to cell signal dropping or switching between towers, this is doing something in the kernel that locks up the whole system until it's done.

$1000 for a description of the root cause and merged pull request that fixes this issue.

cmarshall108, 2020-03-01:
@pd0wm May or may not be the issue, however this call to `subprocess. check_output` will block until the result is returned, this determines what type cellular connection exists. Furthermore, see the comment above the line highlighted below:

https://github.com/commaai/openpilot/blob/2716e37b28c987eed5ce3209e596b2400adb54db/selfdrive/thermald.py#L200

pd0wm, 2020-03-02:
The issue is at a much lower level than just an individual service. Just thermald lagging for a little bit wouldn't cause all the other packets in the system to be delayed too.

AskAlice, 2020-03-29:
I had this happen with tmux open once. I was driving on i70 near 39.712522, -105.304391, which is a major geographic transition, from mountains to metro area.

```
{"event": "alert_add", "alert_type": "canError", "enabled": true, "ctx": {"dongle_id": "6b53237138b1b2c5", "version": "0.7-release", "dirty$
: true, "is_eon": true}}                                                                                                                    
selfdrive/boardd/boardd.cc: missed cycle                                                                                                    
selfdrive/ui/ui.cc: slow frame(221959) time: 238.58  

thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
Result: Parcel(00000000    '....')                                                                                                          
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter                                                                                               
thermald logmessaged logcatd tombstoned uploader ui updated pandad controlsd plannerd loggerd sensord radard calibrationd paramsd camerad mo
deld monitoringd proclogd ubloxd gpsd deleter 

{"event": "upload_failed", "stat": null, "exc": ["ConnectionError(MaxRetryError(\"HTTPSConnectionPool(host='api.commadotai.com', port=443): 
Max retries exceeded with url: /v1.3/6b53237138b1b2c5/upload_url/?path=2020-01-25--21-26-50--73%2Fqlog.bz2 (Caused by NewConnectionError('<$
rllib3.connection.VerifiedHTTPSConnection object at 0x7f91615e10>: Failed to establish a new connection: [Errno 101] Network is unreachable$
))\"))", "Traceback (most recent call last):\n  File \"/data/data/com.termux/files/usr/lib/python3.7/site-packages/urllib3/connection.py\", 
line 157, in _new_conn\n    (self._dns_host, self.port), self.timeout, **extra_kw\n  File \"/data/data/com.termux/files/usr/lib/python3.7/s$
te-packages/urllib3/util/connection.py\", line 84, in create_connection\n    raise err\n  File \"/data/data/com.termux/files/usr/lib/python$
.7/site-packages/urllib3/util/connection.py\", line 74, in create_connection\n    sock.connect(sa)\nOSError: [Errno 101] Network is unreach$
ble\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/data/data/com.t$
[... cut ...]
ges/requests/sessions.py\", line 533, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/data/data/com.termux/files/usr/lib/p$
thon3.7/site-packages/raven/breadcrumbs.py\", line 341, in send\n    resp = real_send(self, request, *args, **kwargs)\n  File \"/data/data/$
om.termux/files/usr/lib/python3.7/site-packages/requests/sessions.py\", line 646, in send\n    r = adapter.send(request, **kwargs)\n  File $
"/data/data/com.termux/files/usr/lib/python3.7/site-packages/requests/adapters.py\", line 516, in send\n    raise ConnectionError(e, reques$

```

pd0wm, 2020-03-30:
That's a traceback from the uploader, probably because the network got interrupted. That's not what's causing the lags.

I grabbed a couple of kernel/android logs around the time of the lag from some of our users.

[lag1.txt](https://github.com/commaai/openpilot/files/4405426/lag1.txt)
lag @ t=90809

[lag2.txt](https://github.com/commaai/openpilot/files/4405428/lag2.txt)
lag @ t=20133

[lag3.txt](https://github.com/commaai/openpilot/files/4405430/lag3.txt)
lag @ 191751

pd0wm, 2020-04-03:
I might have found a way to reproduce the problem. I noticed that a `usleep` sleeps for way longer when you put it the sim card or take it out. Simple c++ program to reproduce:

```c++
#include <iostream>
#include <chrono>
#include <unistd.h>

int main(void){

  while (true){
    long sleep = 1000; // 1ms
    
    auto t_start = std::chrono::high_resolution_clock::now();
    usleep(sleep);
    auto t_end = std::chrono::high_resolution_clock::now();
    double elapsed_time_ms = std::chrono::duration<double, std::milli>(t_end-t_start).count();
    if (elapsed_time_ms > 5.0) {
      std::cout << "Sleep too long: " << elapsed_time_ms << std::endl;
    }
  }

  return 0;
}
```

Do something with the sim card and you see a sleep of over 10ms.

LoganxDev, 2020-04-26:
They all seem to be doing something with the "WifiConfigStore" at the time of the lag.

lag1.txt
`[90805.780590] MAIN 1154 WifiConfigStore - Retrieve network priorities after PNO.`

lag2.txt
`[20127.346106] MAIN 1159 WifiConfigStore -  rewrite network history for "DIRECT-Lq-Car_ffab"-WPA_PSK`

lag3.txt
`[191732.893202] MAIN 1131 WifiConfigStore - Retrieve network priorities after PNO.`

pd0wm, 2020-05-11:
Should be fixed by https://github.com/commaai/openpilot/commit/03a9b546362a5a085ad6261c7901d19f80cfbeb1. Thanks @jyoung8607 

Incoming data from 0.7.5 will confirm this. Will reopen the issue if it's not fixed.

adam-nelson, 2020-06-22:
Could mapping cellular / mobile dead spot help openpilot mitigate and prevent this?

Eg IF in dead spot turn off temporary enable 'airplane' mode

adam-nelson, 2020-06-23:
In my own experience I can't pinpoint this to a geographical spot or dead spot because to mobile coverage area is good / excellent at all time

nabeelr, 2020-08-12:
I can fairly predictably reproduce this error if that matters to anyone. I also have a screen record of it happening with tmux running on my iPhone, that I can pair up to the drive video.

Basically there are a ton of missed cycle messages, then a Receive Buffer full message.

I know a screenshot/video is kind of sacrilege, but this is the best I can do right now.
[image]

I'm willing to do any testing that anyone wants to do considering I can fairly reliably reproduce it, just let me know. That being said, it might take some time, because on this drive after removing the Sim card it fell somewhere, God knows where, in the centre console, in my car. I might need to buy a replacement.

ShaneSmiskol, 2020-09-12:
I've gotten a CAN error out of no where on a road trip I took last month, didn't think anything of it as random errors *have* occurred once or twice in the past. However, it only occurred twice when I was in the middle of nowhere, back to back after restarting the car. Never happened again, the trip was about 700 miles with no issues except those two CAN errors. I don't have a SIM connected, nor comma prime so I'm not sure if this issue would have affected me or not

adeebshihadeh, 2020-09-17:
Fixed by #2194 until a kernel fix is shipped in the next NEOS update.

closed 2020-09-17 by adeebshihadeh