commaai/openpilot #1165: RAV4 19-20 Hybrid: Dashcam Mode - Car Unrecognized

TaiFeng, 2020-02-26:
**Describe the bug**
Ever since upgrading to 0.7.3, Openpilot fails to recognize the RAV4 2019/2020 Hybrid

**How to reproduce or log data**
Example drive where this occurred:
https://my.comma.ai/cabana/?route=e2eec4fedd7f43b6%7C2020-02-25--09-48-38&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fe2eec4fedd7f43b6%2Fc579d25668b1b281e41558cb5826fb1a_2020-02-25--09-48-38&seekTime=113

**Expected behavior**
Similar to 0.7.2, I expect Openpilot to properly detect and interface with my RAV4 2019 Hybrid

**Device/Version information (please complete the following information):**
 - Device: EON Gold
 - Version: 0.7.3 (also tested on my own fork of the same version: https://github.com/TaiFeng/openpilot)
 - Car make/model:  Toyota RAV4 2019 Hybrid XLE 

**Additional context**
@david (.ghost)#7769 on Discord is experiencing the same issue with his RAV4 2020 Hybrid

arne182, 2020-02-26:
https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/values.py#L270 
Causing the issues with White and Grey Pandas.

pd0wm, 2020-02-27:
We merged the fingerprint in 0.7.2, but that caused a lot of other existing cars to break fingerprinting. Some no longer fingerprinted, and there were Corollas fingerprinting as the Rav4. Therefore we removed the ability to fingerprint the 2019 Rav4 Hybrid using can messages. It works fine with the ECU FW method, but you need a black panda or comma 2.

From now on we only merge in new car models for Honda/Toyota using FW based fingerprinting to prevent these issues in the future.

dpanych, 2020-02-27:
> We merged the fingerprint in 0.7.2, but that caused a lot of other existing cars to break fingerprinting. Some no longer fingerprinted, and there were Corollas fingerprinting as the Rav4. Therefore [...]

Will there be a production/release2 fix for rav4s? Or is the best solution to make our fork with the ignore fingerprint value removed?

pd0wm, 2020-02-27:
There will not be production fix. Your best bet is a custom fork or getting hardware which has access to the OBD port.

dpanych, 2020-02-28:
> There will not be production fix. Your best bet is a custom fork or getting hardware which has access to the OBD port.

I was just chatting with a comma two user who has a rav4 as well, bkd0255#3471, he claims his stopped working on 073.

arne182, 2020-02-29:
Might just be missing the firmware version. Just supply a dongle id and it can be updated. Otherwise my fork is open for rav4's even without this fingerprinting problem. @pd0wm why not merge the fingerprints?

routine88, 2020-02-29:
> We merged the fingerprint in 0.7.2, but that caused a lot of other existing cars to break fingerprinting. Some no longer fingerprinted, and there were Corollas fingerprinting as the Rav4. 

My 2019 Camry LE is fingerprinting as  "carFingerprint = "TOYOTA RAV4 2019"",even though all my Camry's FPv2 values are in the Camry section of values.py. Seems like this might be a larger issue than was previously thought? I have a Comma 2.

pd0wm, 2020-03-03:
Because it breaks existing users. When I added the rav4 fingerprint we had cars not fingerprint because they suddenly had two candidates, and some Corollas started fingerprinting as a rav4.

We made a promise to existing giraffe users that we would not break stuff if their car is supported already.

closed 2020-03-03 by pd0wm