commaai/openpilot #1188: @FL pilot 2018 Camry FP as a 2019 RAV4

arne182, 2020-03-03:
dongle_id
9f3e973c64c093f9
FW

<car.capnp:CarParams.CarFw reader (ecu = engine, fwVersion = "\x018966333Q6200\x00\x00\x00\x00", address = 1792, subAddress = 0)>
, 
<car.capnp:CarParams.CarFw reader (ecu = dsu, fwVersion = "8821F0607200    ", address = 1937, subAddress = 0)>
, 
<car.capnp:CarParams.CarFw reader (ecu = esp, fwVersion = "F152606230\x00\x00\x00\x00\x00\x00", address = 1968, subAddress = 0)>
, 
<car.capnp:CarParams.CarFw reader (ecu = eps, fwVersion = "8965B33540\x00\x00\x00\x00\x00\x00", address = 1953, subAddress = 0)>
, 
<car.capnp:CarParams.CarFw reader (ecu = fwdRadar, fwVersion = "8821F0607200    ", address = 1872, subAddress = 15)>
, 
<car.capnp:CarParams.CarFw reader (ecu = fwdCamera, fwVersion = "8646F0605000    ", address = 1872, subAddress = 109)>

FP
{36: 8, 37: 8, 170: 8, 180: 8, 186: 4, 426: 6, 452: 8, 464: 8, 466: 8, 467: 8, 544: 4, 550: 8, 552: 4, 562: 6, 608: 8, 610: 8, 643: 7, 705: 8, 728: 8, 740: 5, 761: 8, 764: 8, 800: 8, 824: 8, 830: 7, 835: 8, 836: 8, 865: 8, 869: 7, 870: 7, 871: 2, 896: 8, 898: 8, 900: 6, 902: 6, 905: 8, 921: 8, 933: 8, 934: 8, 935: 8, 944: 8, 945: 8, 951: 8, 955: 8, 956: 8, 976: 1, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1059: 1, 1076: 8, 1077: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1235: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1594: 8, 1595: 8, 1649: 8, 1745: 8, 1779: 8, 1786: 8, 1787: 8, 1788: 8, 1789: 8}

Has all the signs of a RAV4 2019 but he claims it is a Camry 2018

FLpilot, 2020-03-03:
Hello, arne182 created this for me.  I definitely have a 2019 Toyota Camry.  Thank you.

pd0wm, 2020-03-03:
These are the firmware version:
```    carFw = [
      ( ecu = engine,
        fwVersion = "\x018966333Q6200\x00\x00\x00\x00",
        address = 1792,
        subAddress = 0 ),
      ( ecu = dsu,
        fwVersion = "8821F0607200    ",
        address = 1937,
        subAddress = 0 ),
      ( ecu = esp,
        fwVersion = "F152606230\x00\x00\x00\x00\x00\x00",
        address = 1968,
        subAddress = 0 ),
      ( ecu = eps,
        fwVersion = "8965B33540\x00\x00\x00\x00\x00\x00",
        address = 1953,
        subAddress = 0 ),
      ( ecu = fwdRadar,
        fwVersion = "8821F0607200    ",
        address = 1872,
        subAddress = 15 ),
      ( ecu = fwdCamera,
        fwVersion = "8646F0605000    ",
        address = 1872,
        subAddress = 109 ) ],```

I think I accidentally added a camry into the list of rav4 firmwares. When I made the original database I was basing it of what the car was fingerprinting as using the old style fingerprinting, but that might have been wrong.

pd0wm, 2020-03-03:
Can you confirm this is fixed by: https://github.com/commaai/openpilot/commit/c0b8f4faedccad79f7a13e6951c38a131c0c6ca5?

FLpilot, 2020-03-03:
Yes, that seemed to do the trick.  Fresh install of openpilot on the device, calibration and test drive, and everything works as should.  Thank you for the help and quick troubleshooting.

closed 2020-03-03 by pd0wm