commaai/openpilot #1200: Popular Community Bounty: Toyota no-DSU cars need OP Long

routine88, 2020-03-05:
It bugs us that OP long doesn't work on the Toyota Camry. It's a freaking Camry. Everything is supposed to work on the Camry!

We are banding together to create a community bounty for the betterment of all owners of the best selling sedan in America.....  (And also the CH-R.)

Our pledgers have promised a total of $1,600 and the number is growing by the day (just like the number of Toyota Camrys on the road).

We would like this bounty to be featured on the comma.ai bounty page, and to perhaps get a shout out of some sort to promote hackers to get involved. After all, we're talking about America's favorite sedan.... (And also the CH-R.)

**Expected behavior (OP Long)**
-A chill driving experience

**Actual Behavior (stock TSS-P ACC)**
-PANIC BREAKING 24/7!!!!

Our community bounty pays for a hardware or software solution that enables OP Longitudinal control on no-DSU Camry and CH-R vehicles.

DM @rageser to be added to the registry of pledgers, or interested hackers feel free to inquire.

Thank you all!!!!
@rageser discord

vanillagorillaa, 2020-03-13:
We are consolidating all community bounties to our community bounty page found here, https://community.comma.ai/wiki/index.php/Bounties.

closed 2020-03-13 by vanillagorillaa

routine88, 2020-04-26:
> We are consolidating all community bounties to our community bounty page found here, https://community.comma.ai/wiki/index.php/Bounties.

Please post the above information and the bounty amounts below to the Wiki. I can't seem to gain access to post......

Rageser $500
Andybfield $200
Venu $150
Erich $100
apm $100
aLp $100
Hermosa $100
Djt1283 $100
02trackroad $75
Cferra $50
Jadtec $50
Mubimmer $50
Fchen53 $25

Subtotal: $1600

Community Total = $1600 (hardware or software hack)

Comma bounty = $2000 (software only)