commaai/openpilot #1229: Toyota RAV4 Hybrid 2017 cannot start at 0 mph

junjieshen, 2020-03-11:
Hi,

I installed EON on my Toyota RAV4 Hybrid 2017. Although the this car model is listed as 0 mph for both LON and LAT control, I found that openpilot can only engage after the speed is over 30 mph or so. I was wondering if this is just a typo in the documentation or it's something wrong with my setup.

Thanks,

arne182, 2020-03-11:
Remove DSU from under glove box and it will work from 0

junjieshen, 2020-03-11:
@arne182 Thanks Arne! Will try it out.

pd0wm, 2020-03-11:
Be aware of the fact that disconnecting the DSU disables Automatic Emergency Braking (AEB)!

closed 2020-03-11 by pd0wm

junjieshen, 2020-03-12:
@pd0wm thanks for the info!

junjieshen, 2020-04-01:
Update:
I unplugged the DSU and now it can engage from 0 speed. However, the problem is after that, OP will quickly accelerate to ~28 mph.

For folks want to try this trick, be caution when trying on crowded or bumpy roads, It was a very scary (safe still) experience for me.

@arne182 @pd0wm I was also wondering if there is any way to prevent such acceleration to simply stabilize at a low speed (e.g., 2~3 mph)?

Thanks!

arne182, 2020-04-01:
I still am working on a PR for slow speed engagement preventing the acceleration. It is on my fork if you want to try it so long.