commaai/openpilot #1232: Chrysler Pacifica 2017 being fingerprinted as 2018

arne182, 2020-03-11:
Discord ID: @mbuboltz#4220

dongle_id
4798c754e5a85bac

car doesn't match any fingerprints: {0: {324: 8, 344: 8, 368: 8, 416: 7, 500: 8, 520: 8, 559: 8, 532: 8, 571: 3, 564: 4, 625: 8, 544: 8, 658: 6, 660: 8, 680: 8, 760: 8, 993: 7, 258: 8, 264: 8, 288: 7, 280: 8, 294: 8, 290: 6, 514: 8, 308: 8, 55: 8, 274: 2, 746: 5, 1891: 8, 671: 8, 632: 8, 832: 8, 268: 8, 284: 8, 292: 8, 320: 8, 331: 8, 332: 8, 388: 4, 501: 8, 512: 8, 669: 3, 1251: 8, 448: 6, 736: 8, 729: 5, 257: 5, 792: 8, 384: 8, 863: 8, 672: 8, 800: 8, 1031: 8, 764: 8, 376: 3, 766: 8, 992: 8, 300: 8}, 1: {512: 8, 544: 8, 576: 8, 608: 8, 640: 8, 644: 8, 648: 8, 652: 8, 656: 8, 672: 8, 704: 8, 736: 8, 768: 8, 800: 8, 816: 8, 832: 8, 864: 8, 880: 8, 896: 8, 898: 8, 900: 8, 902: 8, 904: 8, 906: 8, 908: 8}, 2: {324: 8, 344: 8, 368: 8, 416: 7, 500: 8, 520: 8, 559: 8, 532: 8, 571: 3, 564: 4, 625: 8, 544: 8, 658: 6, 660: 8, 680: 8, 760: 8, 993: 7, 258: 8, 264: 8, 288: 7, 280: 8, 294: 8, 290: 6, 514: 8, 308: 8, 55: 8, 274: 2, 746: 5, 1891: 8, 671: 8, 632: 8, 832: 8, 268: 8, 284: 8, 292: 8, 320: 8, 331: 8, 332: 8, 388: 4, 501: 8, 512: 8, 669: 3, 1251: 8, 448: 6, 736: 8, 729: 5, 257: 5, 792: 8, 384: 8, 863: 8, 672: 8, 800: 8, 1031: 8, 764: 8, 376: 3, 766: 8, 992: 8, 300: 8}, 3: {}}

pd0wm, 2020-03-13:
I'm not sure what the issue is. Did you split them out into two separate cars on your fork? Because in openpilot the 2017 and 2018 is the same car.

arne182, 2020-03-13:
You are right. I split them some time ago.

closed 2020-03-13 by arne182