commaai/openpilot #1259: Create soft disengagement when speed is above 90mph

pd0wm, 2020-03-21:


arne182, 2020-03-21:
What is expected?

pd0wm, 2020-03-27:
Some factory ACC allows more than 90mph, e.g. VW in European markets. Since openpilot is not tested at those speeds I want a soft disengage when it goes over 90 mph, e.g. it starts beeping until you take over.

arne182, 2020-03-27:
so you do not want openpilot to work above 90mph even though on the German Roads 105mph is no legal problem. The greater problem is the radar starts to not detect well at such high distances if TR stays at 1.8s. Why not limit the set speed?

jyoung8607, 2020-03-28:
> so you do not want openpilot to work above 90mph even though on the German Roads 105mph is no legal problem.

I don't believe the Comma training dataset has a lot of examples at that speed.

> The greater problem is the radar starts to not detect well at such high distances if TR stays at 1.8s.

VW European market ACC, depending on the vehicle and options, will take a setpoint of 210 kmh / 130 mph. I think the ones that will do 210 either use Bosch LRR, or MRR + MPC sensor fusion to get the necessary look-ahead.

> Why not limit the set speed?

Trying to control the stock ACC setpoint is indirect and hacky at best (virtual button press spamming) and you would still have to Do Something in the interim to alert the user they have done/are doing something unapproved while fixing the setpoint, and this is that Something. The time would be much better spent on OP long control where we do have authority over the setpoint.

arne182, 2020-03-28:
Not the stock acc point but the Openpilot set point. Could this maybe be a community feature to go faster.

PR #1295 (2020-03-31): High speed alert
Fixes #1259
closed 2020-03-31 by pd0wm