commaai/openpilot #1270: OpenPilot (engaged) sporadically creeps forward slowly at stopped traffic while at a red light

dpanych, 2020-03-24:
**Describe the bug**
While OP is engaged, when coming to a complete stop (like a red light), OP will first stop at a normal distance, but then it will continue to creep toward the stopped car (very slowly). 

**How to reproduce or log data**
I am unable to create this as it seems to be random. For futures sake, I will save cabana links when it occurs again.
https://my.comma.ai/cabana/?route=7e34a988419b5307%7C2020-03-19--19-19-14&exp=1616186697&sig=GOzZLgLrNbOUJ9FTgnAQF0HTbhe34bN28UfWwi6UJo4%3D&max=16&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F7e34a988419b5307%2F1dac8843e1a4c769857cd329d8baad59_2020-03-19--19-19-14 (Look at 466.405)

**Expected behavior**
The car should stop and stay stopped until traffic starts moving again.

**Device/Version information (please complete the following information):**
 - Device: EON GOLD
 - Version: 0.7.4 staging-devel
 - Car make/model: 2020 RAV4 Hybrid XLE

arne182, 2020-03-26:
Have seen this on my fork users. 
I have increased this line from longcontrol form 0.5 to
BRAKE_STOPPING_TARGET = 0.75  # apply at least this amount of brake to maintain the vehicle stationary
Giving 25% more braking at standstill.

arne182, 2020-03-26:
I still think we should include lead car distance into longcontrol to avoid this having both the rate and the amount dynamically change based on lead car behaviour.

pd0wm, 2020-03-27:
How much braking does the stock system send at standstill?

We send an acceleration to the car, 0.5 braking maps to -1.5 m/s^2. Theoretically everything smaller than zero should keep the car stopped, since they have a lower level controller that should keep the brakes applied.

arne182, 2020-03-27:
Yes 0.5 -1.5m/s^2 should keep the car at a standstill but it does not for the TSS2 RAV4 and hybrid. Having -2.0 to -2.5m/s^2, does keep the car from creeping. The funny part is that vEgo is 0 but lead distance is decreasing.

arne182, 2020-03-27:
[image]

dpanych, 2020-03-28:
Seems to have happened again @ 365 behind the Subaru - seemed like it was creeping (even my wife noticed) but then light when green. 
https://my.comma.ai/cabana/?route=7e34a988419b5307%7C2020-03-27--01-42-29&exp=1616910719&sig=N2xA%2Bp6tIw%2Fssr4uZfj%2Bng3rWFd%2BgnZw9FwOkVOQ%2BVg%3D&max=12&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F7e34a988419b5307%2F47759ad23f7c5ad63a1f886861e6eb3c_2020-03-27--01-42-29

pd0wm, 2020-03-28:
@arne182 , is that a screenshot from the stock system?

arne182, 2020-03-28:
No. From the open pilot one. Let me ask for a stock drive. BRB

arne182, 2020-03-30:
Stock system uses -3.5m/s and then goes up to 0.
[image]
I also see a -1.28m/s constantly held without creeping.
in @dpanych drive I see the RELEASE_STANDSTILL is not set to 0 for the whole standstill time
[image]

dpanych, 2020-03-30:
Here's another example:
614.72 @ https://my.comma.ai/cabana/?route=7e34a988419b5307%7C2020-03-29--19-02-02&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F7e34a988419b5307%2F06f794ca602d8062e46bc2fc29da8827_2020-03-29--19-02-02&seekTime=61

Tommytech9, 2020-05-22:
I too have this issue on 2020 Corolla Hybrid on 0.7.5. Can this car be tweaked as well?

directedbit, 2020-06-16:
Had the same issue on 2020 RAV4 TSS2

Taik, 2020-07-22:
I can confirm this is still happening on my 2020 RAV4 TSS2 (077 release2).

Volbard, 2020-08-02:
I'm also seeing this.

Device/Version information:
    Device: Comma Two
    Dongle ID: 93cb227004373575
    Route: 93cb227004373575|2020-08-01--13-43-51
    Version: release2 0.7.7
    Car make/model: 2020 Rav4 Hybrid

Worst case is 13:57 on the route.

srieg, 2020-08-27:
I have this issue on my 2020 Highlander Hybrid as well

directedbit, 2020-10-13:
@pd0wm - should we wait for this to get fixed or just use @arne182's PR? It's a little unnerving having the car creep forward towards other stationary vehicles

C0mpl3t3N00b, 2020-11-02:
Can confirm this is still happening even on master-ci as experienced today.

https://my.comma.ai/cabana/?route=bdad50b8a1f9fa8e%7C2020-11-01--16-47-58&exp=1635827347&sig=huumgNlfxiHDVminmiuY9uGtn%2FL3BO%2FrcSWvX5BBpg4%3D&max=18&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fbdad50b8a1f9fa8e%2F6cd92beecc597e5fb5f53b068f0c3f7a_2020-11-01--16-47-58

RightTrash, 2020-11-27:
Have a 2020 Rav4 Hybrid XLE with comma 2, 7.10 and at every stop it does on it's own, it comes to the stop, then suddenly creeps and stops again.  Sometimes it then creeps more so, the second creeping doesn't always occur, there have been numerous times I've disengaged it because I felt I was creeping too close to the stopped car.  
The only time I would say something seemed apparent, perhaps relating, was at an odd 4 lane stop light which was mid sharp bend with a 2 lane one way road, where both roads are intersecting mid sharp bends on a hill also, the car in front of me was at an angle somewhat...

Hope to see this issue/bug worked out, as it seems sketchy in various ways, always ready to disengage in those moments.

adeebshihadeh, 2020-12-17:
This should be fixed by d93f07f5e6d550731b024521a2bda33325bcc640

closed 2020-12-17 by adeebshihadeh