commaai/openpilot #1278: ACC can by bypassed on Chrysler

mbuboltz1024, 2020-03-26:
ACC can by bypassed on Chrysler. Openpilot allows engagement when ACC is disabled.

How to reproduce or log data
Press the cruise on/off button instead of ACC on/off button. ACC will be disabled and the car will switch to non-adaptive cruise control. Press set to engage. Openpilot will control steering, but ACC will not function.

Expected behavior
Openpilot should not be allowed to be engaged without ACC.

Device/Version information:

Device: EON Gold
Version: 0.7.4
Car make/model: 2017 Chrysler Pacifica (Gas)

adhintz, 2020-03-26:
Can you please share a drive where this happens? Feel free to send it directly to me if you don't feel like publicly posting it. Please use Cabana's "Copy Share Link" button and paste the address.

mbuboltz1024, 2020-03-26:
Here is a drive when it was using Chrysler ACC. One without ACC to come soon.
https://my.comma.ai/cabana/?route=4798c754e5a85bac%7C2020-03-24--13-51-46&exp=1616779094&sig=PTFW4mMIdwInI6wzcteeZAbJ37oDhJGs0Fb7dAlIMfg%3D&max=5&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F4798c754e5a85bac%2F430ed75c63c8d3c7acd4d80e15bbcb96_2020-03-24--13-51-46

mbuboltz1024, 2020-03-26:
Here is my drive where OpenPilot allows you to use regular cruise where it does not brake. I tried to show some instances where it would get too close to car in front of me. Didn't want to get to close lol didn't want to piss anyone off.
https://my.comma.ai/cabana/?route=4798c754e5a85bac%7C2020-03-26--13-25-54&exp=1616783949&sig=isbxixOIrG26Svik9sJZa1WswbVK2YLa6Ep8Aee7a28%3D&max=6&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F4798c754e5a85bac%2F208a9846849f747fadf3d268ee6ff8d5_2020-03-26--13-25-54

mbuboltz1024, 2020-03-26:
OpenPilot has never controlled braking for my vehicle, it has always relied on factory ACC. I hope it is supported in the future so it can slow down for curves/turns.

adeebshihadeh, 2020-06-16:
#1708 adds the support code necessary for this. All the alerting and state will be handled if you set the `cruiseState.nonAdaptive` flag.

closed 2021-01-06 by adeebshihadeh