commaai/openpilot #1294: Add support for Hyundai radar parsing -- $200 bounty

geohot, 2020-03-30:
Put here: https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/radar_interface.py

Fine if it's only on cars with SCC.

Olshansk, 2020-04-07:
I saw this and looked at a few of the other RadarInterface implementations. There are a few commonalities, but the business logic seems to really vary.

I honestly don't think I have enough context to solve this, but I'm curious if you could at least give a quick pointer of how one would go about doing this? Do I need raw data to analyze as a test set?

TK211X, 2020-05-26:
Can the reward be a Comma 2 instead of $200?

geohot, 2020-06-26:
Increased bounty due to difficultly of getting radar points. I don't obviously see them on bus 1, the camera points don't count.

TK211X, 2020-06-27:
Inspection of a disconnect event from MFC demonstrates that 3 messages continue to be created by the Radar ECU.

These are `0x5ED` `0x5EE` `0x5EF`, they appear to serve the purpose of signaling the presence of the Radar ECU on LCAN and only contain a Counter and Checksum on all three messages.

The fallout of the messages on LCAN during MFC disconnect is as follows

43 messages ceased transmission @ .003 Mark / 0.030 FREQ

2 messages ceased transmission @ .013 Mark / 0.030 FREQ

7 messages ceased transmission @ .023 Mark / 0.030 FREQ

3 messages continued to transmit. / 0.050 FREQ

Next course of action is to attempt to trigger the Radar ECU into spitting out radar data.

Most LCAN status messages are very simple and just have a counter. It's very possible you could simply set up a script that goes through every ID on LCAN and generates a message for a few seconds containing a counter in each byte. Several could trigger a set of radar tracks to be output.

TK211X, 2020-06-27:
I can't seem to understand the difference in models that output Radar tracks onto LCAN and the Sonata where they they decided not to do this for whatever reason.

J-Rojas, 2021-04-07:
Has there been any progress on retrieving the radar tracks for Genesis and Niro which have an LCAN bus? I believe these cars have a variant of this Aptiv/Delphi ESR 2.5 radar: https://autonomoustuff.com/products/aptiv-esr-2-5-24v

jonjander, 2021-09-15:
@gregjhogan @pd0wm 
I used **hyundai_kia_mando_front_radar.dbc** on data from my Kia Niro. The messages didn't match, but I found something that looks like dRel on address 0x238. Can you share some sample data that I can use for comparison?

closed 2021-09-15 by gregjhogan