commaai/openpilot #1330: Find out how to control EPB

pd0wm, 2020-04-07:
Whenever we trigger a forced disengagement, e.g. when the driver leaves the vehicle, and the vehicle is stopped we want to trigger EPB to keep the car stopped.

pd0wm, 2020-04-14:
Checked with Techstream, and can't control it through there

pd0wm, 2020-05-11:
Looked through all the can data when the stock system applies EPB and couldn't find anything. For the next experiment I'm going to make openpilot go through the same stop, resume cruise states as the stock system. Maybe EPB will automatically apply when opening the door when going through those states.

xps-genesis, 2020-05-13:
EPB control is integrated in ESP for most vehicles. ESP will set EPB at standstill with ACC engaged, when there is any possible condition that may cause safety issues-
1.) driver trying to exit the vehicle,(open door or seat belt)
2.) ESP overheating due to long standstill hold in ACC
3.) any failure in ESP that will lead to brake torque release
4.) ESP unable to keep the vehicle at standstill, multiple creep events when ACC is commanding standstill.(possible faded brake pads or hydraulic leak?

pd0wm, 2020-05-13:
@xps-genesis The goal is to have openpilot do the same thing when the driver leaves the vehicle. Unfortunately we don't go through the same standstil logic because that would require you to press resume. I was hoping there is a way to request the parking brake to be set over can by openpilot.

LtAhmed776, 2021-08-05:
Good moornig

CineXWorm, 2022-01-10:
> @xps-genesis The goal is to have openpilot do the same thing when the driver leaves the vehicle. Unfortunately we don't go through the same standstil logic because that would require you to press resume. I was hoping there is a way to request the parking brake to be set over can by openpilot.

For what makes are you looking for this info?
For the VW PQ platform this can be done by sending the seatbelt signal or driver door signal.
The last option is spoofing ignition.

closed 2022-12-30 by adeebshihadeh