commaai/openpilot #1349: Communication Error on sim with carla

j-o-d-o, 2020-04-10:
**Describe the bug**
Followed the instructions to get openpilot running with Carla. Once the bridge starts, opendbc seems to have issues with the counter of the signals:
```bash
...
0x17C COUNTER FAIL 2 -- 3 vs 3
0x17C COUNTER FAIL 2 -- 0 vs 0
0x17C COUNTER FAIL 2 -- 1 vs 1
0x17C COUNTER FAIL 2 -- 2 vs 2
...
```
As far as I can tell, the signal counter does not increase. Yet, when creating the can messages the idx does increase.

Also os.read() seems to have issues reading from stdin as the input size to input does not have the proper input size (but I assume this is a follow up problem of the one above?):
```
Traceback (most recent call last):
  File "/home/jodo/openpilot/selfdrive/modeld/runners/keras_runner.py", line 57, in <module>
    run_loop(m)
  File "/home/jodo/openpilot/selfdrive/modeld/runners/keras_runner.py", line 32, in run_loop
    ret = m.predict_on_batch(idata)
...
ValueError: Cannot reshape a tensor with 0 elements to shape [1,12,128,256] (393216 elements) for 'model/reshape/Reshape' (op: 'Reshape') with input shapes: [1,0], [4] and with input tensors computed as partial shapes: input[1] = [1,12,128,256].
```

**How to reproduce or log data**
Run openpilot with Carla as described [here](https://github.com/commaai/openpilot/tree/master/tools/sim).

**Device/Version information (please complete the following information):**
 - Device: Linux Mint 19.2
 - Version: latest master (85a09b3113fe45b5feee7c1c421e2c43cb2ac03a)
 - Car make/model: carla sim

ZwX1616, 2020-04-30:
Which version of tensorflow were you running?

johschmitz, 2020-06-16:
Having the same issue with Mint 19.3
@pd0wm: Tensorflow version is 2.2.0 according to Pipfile.lock
Could it be the wrong CUDA or NVIDIA driver version or something like that?
Maybe you could tell us the versions you are using in-house.

ed2k, 2020-07-12:
remove this line https://github.com/commaai/openpilot/blob/master/tools/sim/lib/can.py#L41
`msg.append(packer.make_can_msg("POWERTRAIN_DATA", 0, {}, idx))`

closed 2020-09-10 by pd0wm

ucolchen, 2020-12-25:
[image]