commaai/openpilot #1358: No module named 'cereal.messaging.messaging_pyx

tom-doerr, 2020-04-13:
**Describe the bug**
I'm trying to run Openpilot in Carla as described here: https://github.com/commaai/openpilot/tree/master/tools/sim
When I execute
```
PASSIVE=0 NOBOARD=1 ./manager.py 
```
in `selfdrive/` I get the following error message:
```
scons: building terminated because of errors.
Traceback (most recent call last):
  File "./manager.py", line 135, in <module>
    add_logentries_handler(cloudlog)
  File "/home/tom/git/openpilot/selfdrive/swaglog.py", line 39, in add_logentries_handler
    from selfdrive.logmessaged import get_le_handler
  File "/home/tom/git/openpilot/selfdrive/logmessaged.py", line 4, in <module>
    import cereal.messaging as messaging
  File "/home/tom/git/openpilot/cereal/messaging/__init__.py", line 2, in <module>
    from .messaging_pyx import Context, Poller, SubSocket, PubSocket  # pylint: disable=no-name-in-module, import-error
ModuleNotFoundError: No module named 'cereal.messaging.messaging_pyx'
```
Any idea what the issue could be?

pd0wm, 2020-04-13:
Scons failed building. What do you get if run `scons -j4` manually in the openpilot folder?

tom-doerr, 2020-04-13:
Thanks for the quick reply! 
With 60f6e12 the error stays the same, just did a pull.
This happens when I execute `scons -j4`:
```
(venv) ➜  openpilot git:(master) ✗ scons -j4                            
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
capnpc cereal/car.capnp cereal/log.capnp --src-prefix=cereal -o c:cereal/gen/c/
sh: 1: capnpc: not found
capnpc cereal/car.capnp cereal/log.capnp --src-prefix=cereal -o c++:cereal/gen/cpp/
sh: 1: capnpc: not found
clang++ -o cereal/messaging/bridge -Wl,-rpath=/home/tom/git/openpilot/phonelibs/capnp-cpp/x64/lib -Wl,-rpath=/home/tom/git/openpilot/phonelibs/zmq/x64/lib -Wl,-rpath=/home/tom/git/openpilot/external/tensorflow/lib -Wl,-rpath=/home/tom/git/openpilot/cereal -Wl,-rpath=/home/tom/git/openpilot/selfdrive/common cereal/messaging/bridge.o -Lphonelibs/capnp-cpp/x64/lib -Lphonelibs/capnp-c/x64/lib -Lphonelibs/snpe/x86_64-linux-clang -Lphonelibs/zmq/x64/lib -Lphonelibs/libyuv/x64/lib -Lexternal/zmq/lib -Lexternal/tensorflow/lib -Lcereal -Lselfdrive/common -L/usr/lib -L/usr/local/lib -Lcereal -Lselfdrive/common -Lphonelibs -Lphonelibs/zmq/aarch64-linux/lib -Lphonelibs/capnp-cpp/aarch64-linux/lib cereal/libmessaging.a -lzmq
scons: *** [cereal/gen/c/car.capnp.c] Error 127
scons: *** [cereal/gen/cpp/car.capnp.c++] Error 127
cd cereal/messaging && python3 messaging_pyx_setup.py build_ext --inplace
Compiling messaging_pyx.pyx because it changed.
[1/1] Cythonizing messaging_pyx.pyx
/home/tom/anaconda3/lib/python3.7/distutils/dist.py:274: UserWarning: Unknown distribution option: 'nthreads'
  warnings.warn(msg)
running build_ext
building 'messaging_pyx' extension
creating build
creating build/temp.linux-x86_64-3.7
gcc -pthread -B /home/tom/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/tom/git/openpilot/cereal/messaging -I/home/tom/git/openpilot/venv/include -I/home/tom/anaconda3/include/python3.7m -c messaging_pyx.cpp -o build/temp.linux-x86_64-3.7/messaging_pyx.o -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
g++ -pthread -shared -B /home/tom/anaconda3/compiler_compat -L/home/tom/anaconda3/lib -Wl,-rpath=/home/tom/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/messaging_pyx.o /home/tom/git/openpilot/cereal/messaging/../libmessaging.a -lzmq -o /home/tom/git/openpilot/cereal/messaging/messaging_pyx.so
scons: building terminated because of errors.

```

pd0wm, 2020-04-13:
make sure you add `external/bin` to your path. It can't find capnpc

j-o-d-o, 2020-04-14:
Had the same error. I didn't see the `external/bin` at first, so I ran `./cereal/install_capnp.sh`, should also do it. It installs capnp to your `/usr/local/bin` and `/usr/local/lib`. </br>
Or adding `source /PATH/TO/openpilot/tools/openpilot_env.sh` would also take care of adding capnp to your PATH. </br>
But adding `external/bin` to PATH seems the cleanest.

pd0wm, 2020-04-15:
@tom-doerr I assume setting the path fixed your issue. Closing this.

closed 2020-04-15 by pd0wm