commaai/openpilot #1364: Openpilot not working with carla, no response when keys are pressed

itsyashkothari, 2020-04-14:
OS: Ubuntu 18.04
Followed the procedure given at tools/sim/README.md

Carla is getting started with no error.
After running manager.py i get following error but it continues running. 
loggerd and modeld are shown in red whereas rest are in green.
```
loggerd: selfdrive/loggerd/loggerd.cc:631: int main(int, char **): Assertion `err == 0' failed.
```

After running bridge.py from sim, when i press 1, in manager.py terminal i get 
```
{"event": "alert_add", "alert_type": "commIssueNoEntry", "enabled": false, "ctx": {"dongle_id": "e8c37a13d6b5c170", "version": "0.7.5", "dirty": true, "is_eon": true}}
```

ZwX1616, 2020-04-30:
Looks like the model is still running with the SNPE API (should use the tensorflow runner on PC) which throws an error and prevents op from engaging. Have you tried rebuliding openpilot?

pd0wm, 2020-05-11:
Closing this issue. Feel free to open a new issue if you still have issues.

closed 2020-05-11 by pd0wm

robelkebede, 2020-06-05:
i have tried rebuilding open pilot and it is still giving me the above error