commaai/openpilot #1401: Cannot build OpenPilot with "scons" in Ubuntu 18.x

NickJohn547745, 2020-04-20:
**Describe the bug**
When following the guide to deploy OpenPilot on Ubuntu 18.x, the steps lead to building the project with "scons use_webcam=1." However, when doing this it simple throws errors as the function doesn't expect an encoding argument to be passed.

**How to reproduce or log data**
Follow the detailed steps to deploy on Ubuntu 18.x

**Expected behavior**
The project builds completely and the modules can then be used.

**Device/Version information (please complete the following information):**
Device: HP Latitude
OS: Ubuntu 18.x
Python: 2.7.17 & 3.6.9

pd0wm, 2020-04-20:
Can you post the output?

Make sure you install scons through the python 3 pip. The one from the ubuntu repos is python 2 based and doesn't work.

NickJohn547745, 2020-04-20:
@pd0wm You're exactly correct! I ended up taking the long route and manually uninstalled the older version. I also had to declare in the SCons executable to use python 3.6.

closed 2020-04-20 by NickJohn547745

LIHaoran-good, 2021-05-15:
Hi , I am trying to run openpilot with webcam on PC now, I refer to this guideline that Build openpilot for webcam, when entering the compile instruction "USE_WEBCAM=1 scons -j$(nproc)", report the error "scons: *** Import of non-existent variable "libkf" ", I don't know what the issue is, do you know the problem is ?
In addition, can I add your contact information to communicate with you.
Thanks very much.

Operating system: [Ubuntu 20.04]

jakubLangr, 2021-05-16:
Hi @LIHaoran-good I am getting the same issue. Did you manage to solve your error with `libkf`? If so, could you post it here or on discord? Thanks!

LIHaoran-good, 2021-05-17:
Hi @jakubLangr ,  I am trying to solve the error with libkf now,  but the root cause and solution have not been found. I will post it here if it is solved,  meanwhile, please tell me if you solved it, and we can communicate about it together. Thanks very much.

LIHaoran-good, 2021-05-17:
Hi @jakubLangr ，I have a question to ask you , I use CAN card instead of Panda to communicate with vehicle, Do you use Panda to communicate with vehicle