commaai/openpilot #1423: Joystickd readme in tools outdated

codename224, 2020-04-26:
Joystickd in tools does not seem to function, and it's readme instructions are out of date. 

When running all three scripts with no visible console errors and all dependencies working correctly, joystickd.py does not move the steering wheel. 

The readme instructions are also out of date. For example, you do not need to manually set ALL_OUTPUT anymore.

pd0wm, 2020-04-27:
Everything should still work, I used it pretty recently. I updated the readme with some new instructions. Can you get it to work with those instructions?

jyoung8607, 2020-04-27:
Another thing that will stop joystick use (and MOCK boardd, and a lot of the Panda Python example scripts) is running firmware built for EON, including the default/signed shipped FW from Comma. In short, a Panda flashed for actual driving with openpilot won't work for any of these tools.

Firmware built with the EON option expects a keepalive from OP boardd, and the missing keepalive kicks Panda back to NOOUTPUT within a couple seconds after the tool sets an output mode. 

It's very easy to run into this, and hard for a newbie to figure out because the failure is totally silent, and it's something of an annoyance to reflash Panda if your usual development environment is virtualized.

I ran into this the hard way recently, but had not submitted a PR to fix it because I'm not sure of a workable approach that Comma would find acceptable. I don't like having to flash back and forth, and I'd especially like to be able to use the Panda tools from a vehicle-installed EON/C2. For my own development purposes, I hacked around it by setting the heartbeat timeout to an entire day.

Alternatives are to add a USB command to disable (or lengthen) keepalives, or to augment all the various development tools to send the heartbeat commands periodically. Or, even if no automatic workaround is acceptable, we could at least have the dev tools detect EON firmware and bail out with an informative notice.

AhadCove, 2020-04-27:
I tried to get joystickd to work as well for some time to no avail. But what you mentioned @jyoung8607 sounds pretty much the same issue I was having, it was silently failing, but the code was still running and looked as if it was working, but nothing would happen.

If you have a simple guide to what you did to get past this it would be extremebly helpful. Did you create your own panda firmware with the heartbeat or eon firmware?

jyoung8607, 2020-04-28:
I think if you follow Willem's updated instructions, it's more likely to work for officially supported cars since it's using the new `boardd` that emits keepalives. I have not tested this personally with an EON or Comma signed build.

For the old instructions using `boardd_old.py`, with the default Panda firmware, I think you would have trouble with missing keepalives on EON builds, and also you can't go into ALL_OUTPUT mode without DEBUG firmware built from source. You'd also have trouble if your car's safety mode is in the [ALLOW_DEBUG group](https://github.com/commaai/panda/blob/master/board/safety.h#L200-L223) as not yet officially supported.

You should retry with Willem's updated instructions. If that doesn't work, you should build the Panda firmware yourself from source, ideally on Ubuntu bare metal (virtualization makes it difficult to carry out the reflash step) so all the DEBUG stuff is enabled for you.

closed 2020-05-11 by pd0wm