commaai/openpilot #1424: Intermittent "Car Unrecognized"

Rod1777, 2020-04-26:
**Describe the bug**
I nearly always get "Car Unrecognized" when I turn on the ignition. Turning the ignition off and on again usually corrects the issue then the drive is normal.

I've tried a different ethernet cable but no improvement.

**How to reproduce or log data**
I am able to reproduce this using fw_versions.py with this procedure:

cd /data/openpilot/selfdrive/boardd
pkill ./manager.py
./boardd
// Then in another session/window:
cd /data/openpilot/selfdrive/car/
python fw_versions.py
// Start Car when message shows getting Vin
// then run fw_versions.py again
python fw_versions.py

The attached file shows the results. The first script run yields no data. The second run immediately afterwards returns the correct serial numbers.

**Device/Version information (please complete the following information):**
 - Device: Comma two
 - Version: 0.7.4
 - Car make/model: Honda CR-V 2019 (Australia)

**Additional context**
[OpenPilotDebug.txt](https://github.com/commaai/openpilot/files/4534635/OpenPilotDebug.txt)

pd0wm, 2020-04-26:
What is your dongle id? I'd also look at some of your driving data.

Rod1777, 2020-04-26:
b17f63924b1f2f94

Rod1777, 2020-04-27:
I'm wondering if the fingerprint 2.0 code runs too soon after ignition on.
I am prepared to make code changes and test.

pd0wm, 2020-04-27:
The FW versions for your car were not in the fingerprinting 2.0 database. It was falling back to old style fingerprinting causing it to fail sometimes.

I added your FW versions: https://github.com/commaai/openpilot/commit/d837b68a764d4a1e88893effd022f208d4488a07. So your problem should be fixed when that ships to release.

closed 2020-04-27 by pd0wm

Rod1777, 2020-04-28:
I have tested the changes, it does appear to have corrected the issue. It is noticeably faster to find the car also.