commaai/openpilot #1467: name: boardd  >>> ./boardd <<< signal 7 (SIGBUS), code 1 (BUS_ADRALN)

geohot, 2020-05-04:
@deanlee I think MessageReader broke boardd. BUS_ADRALN is an alignment issue.

The super minor perf improvements are not worth it!

geohot, 2020-05-04:
For now, I have reverted the commit.

https://github.com/commaai/openpilot/commit/d5ca841b0f18d2e707eb1f73afac842ede9be7e8

9 people ran into this issue 12 times. Posting the full sentry report below

geohot, 2020-05-04:
'*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
CM Version: '13.0-20191002-UNOFFICIAL-Sultan-oneplus3'
Build fingerprint: 'OnePlus/OnePlus3/OnePlus3:6.0.1/MMB29M/362280:user/release-keys'
Revision: '0'
ABI: 'arm64'
pid: 2948, tid: 2975, name: boardd  >>> ./boardd <<<
signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0x4101ce0f374bc6
    x0   0000000000000040  x1   0000000000000004  x2   0000000000000008  x3   000000000000002d
    x4   0000000000000003  x5   0000000000000004  x6   0000000000000005  x7   0000007f84852000
    x8   0000007f84852040  x9   000000000002da80  x10  0000000000002b20  x11  0000000000002c00
    x12  0000000000000000  x13  0000007f84892000  x14  0000000000000000  x15  0000007f84884000
    x16  0000007f84895f80  x17  0000007f84884000  x18  0000007f84896020  x19  b100000000000000
    x20  003b261912000004  x21  bb0002e3b0000001  x22  4c008a208e00002e  x23  4f0000004900002e
    x24  4fffffff9400002a  x25  0000000000002710  x26  0000007f84812000  x27  0000000000000000
    x28  0000000000001000  x29  a800f0150262b517  x30  384101ce0f374bc6
    sp   0000007f8313d240  pc   004101ce0f374bc6  pstate 0000000080000000
    v0   ffffff80ffffffe00000007f8313d140  v1   0000007f8313d1600000007f8313d1b0
    v2   b100000000000000003b261912000004  v3   384101ce0f374bc6a800f0150262b517
    v4   63632e646472616f622f646472616f62  v5   222c3135383a226f6e656e696c222c22
    v6   6f65676970223a22656d616e636e7566  v7   3733322e313438393036383835313a22
    v8   00000000000000000000000000000000  v9   00000000000000000000000000000000
    v10  00000000000000000000000000000000  v11  00000000000000000000000000000000
    v12  00000000000000000000000000000000  v13  00000000000000000000000000000000
    v14  00000000000000000000000000000000  v15  00000000000000000000000000000000
    v16  000000000000000041cdcd6500000000  v17  000000000000000041d7ac10cc4f2cd2
    v18  00000000000000004021051eb851eb85  v19  00000000000000004020000000000000
    v20  0000000000000000000000004eb22efa  v21  00000000000000004059000000000000
    v22  0000000000000000000000000000ff4a  v23  00000000000000000000000000ff4aba
    v24  00000000000000003ff0000000000000  v25  00000000000000003ff0000000000000
    v26  000000000000000000000000b6d4c832  v27  000000000000000000000000bbc618cf
    v28  000000000000000000000000cb673761  v29  000000000000000000000000508de1a4
    v30  0000000000000000000000007b41653b  v31  00000000000000000000000086e6c55a
    fpsr 00000010  fpcr 00000000

backtrace:
    #00 pc 004101ce0f374bc6  <unknown>
    #01 pc 384101ce0f374bc2  <unknown>

stack:
         0000007f8313d1c0  0000007f8313d230  [stack:2975]
         0000007f8313d1c8  0000007f848a2068  /system/comma/usr/lib/libusb-1.0.so.0.1.0 (libusb_control_transfer+492)
         0000007f8313d1d0  0000007f8313d200  [stack:2975]
         0000007f8313d1d8  000000018da04e74
         0000007f8313d1e0  0000000000001000
         0000007f8313d1e8  0000000000000000
         0000007f8313d1f0  0000007f84812000  /system/lib64/libc.so (strpbrk+32)
         0000007f8313d1f8  0000000000002710
         0000007f8313d200  4fffffff9400002a
         0000007f8313d208  4f0000004900002e
         0000007f8313d210  4c008a208e00002e
         0000007f8313d218  bb0002e3b0000001
         0000007f8313d220  003b261912000004
         0000007f8313d228  b100000000000000
         0000007f8313d230  a800f0150262b517
         0000007f8313d238  384101ce0f374bc6
    #00  0000007f8313d240  0000007f83b46040  [anon:libc_malloc]
         ........  ........
    #01  0000007f8313d240  0000007f83b46040  [anon:libc_malloc]
         0000007f8313d248  0000007f83b46050  [anon:libc_malloc]
         0000007f8313d250  0000000000000000
         0000007f8313d258  0000000000000000
         0000007f8313d260  0000000000000000
         0000007f8313d268  0000000000000000
         0000007f8313d270  0000000000000000
         0000007f8313d278  0000000000000000
         0000007f8313d280  0000000000000000
         0000007f8313d288  0000000000000000
         0000007f8313d290  0000000000000000
         0000007f8313d298  0000000000000000
         0000007f8313d2a0  0000000000000000
         0000007f8313d2a8  0000000000000000
         0000007f8313d2b0  0000000000000000
         0000007f8313d2b8  0000000000000000

geohot, 2020-05-04:
Before merging this again, we need a test that repros this alignment issue.

I suspect the problem might be "buf = (capnp::word *)malloc(words * sizeof(capnp::word));" not ensuring that buf is word aligned.

deanlee, 2020-05-04:
don't know why....
I'm so disappointed with myself,wasting too many time to trying to optimize the least significant thing.

geohot, 2020-05-04:
It's okay, I didn't catch the bug on reading, and CI didn't catch it either.

It's unclear it's related to the perf refactor, but considering it's BUS_ADRALN in boardd, this commit seemed highly suspect.

Let's see if we can repro and write a test, that's how we get better.

deanlee, 2020-05-04:
there are two possible problems,one is the if statement, another is mabe should using calloc or memset to initializes all bits to zero.I'm reading capnproto's source code,try to find out where the problem is

geohot, 2020-05-04:
I don't think it's a zeroing thing. On ARM (but not x86!), misaligned accesses can cause exceptions.

geohot, 2020-05-04:
This doesn't repro it, though perhaps with tweaks it can?

`#include <stdio.h>
#include <stdlib.h>
#include <capnp/serialize.h>
#include "cereal/gen/cpp/log.capnp.h"

int main() {
  capnp::MallocMessageBuilder msg;
  auto event = msg.initRoot<cereal::Event>();
  event.setLogMonoTime(22);
  auto words = capnp::messageToFlatArray(msg);
  auto bytes = words.asBytes();

  int size = bytes.size();
  char *data = (char *)malloc(size+100);
  data++;

  memcpy(data, bytes.begin(), bytes.size());

  capnp::FlatArrayMessageReader *cmsg = new capnp::FlatArrayMessageReader(kj::ArrayPtr<capnp::word>((capnp::word *)data, size / sizeof(capnp::word)));
  cereal::Event::Reader eventr = cmsg->getRoot<cereal::Event>();

  int a = eventr.getLogMonoTime();
  printf("%d\n", a);
}
`

deanlee, 2020-05-04:
good testing program.I'm trying repro that issue based on it.

deanlee, 2020-05-05:
Still have not reproduced this problem.i'll continue trying.

btw,pycapnp has the same way of implementation(if (<uintptr_t>ptr) % 8 != 0:), does it caused any problems before?

https://github.com/capnproto/pycapnp/blob/9aa3d981f1070a8cc96a58b8f6471197df3ed2f4/capnp/lib/capnp.pyx#L3793-L3798

geohot, 2020-05-05:
No, it hasn't. Though I suspect %8 alignment is what is needed, not %2, even though it's a word.

deanlee, 2020-05-06:
can't repro it no matter how I adjust the memory address and message body.Is it possible that the problem is elsewhere such as libusb_control_transfer?

> stack:
0000007f8313d1c0 0000007f8313d230 [stack:2975]
0000007f8313d1c8 0000007f848a2068 /system/comma/usr/lib/libusb-1.0.so.0.1.0 (libusb_control_transfer+492)

deanlee, 2020-05-07:
still can't repro it no matter how i tweek it. Is it possible that it's a problem somewhere else?

geohot, 2020-05-08:
It's possible it's something else assuming alignment of that packet. I propose we bring MessageReader back and merge it, but just have it always make a copy using the exact code we have cut and pasted everywhere. Then we have a good base, and can think about performance later.

deanlee, 2020-05-08:
forget about messagereader,I'm wrting a new Pub&Sub socket manager in cereal.will put FlatArrayMessageReader&Event (with malloced cache per endpoint) in SubMaster , and put messageToFlatArray in PubMaster's send function.hopefully that will reduce many redundant code

closed 2020-05-11 by pd0wm

deanlee, 2020-06-17:
Did this issue happen again after replaced with SubMaster ? @geohot  @pd0wm