commaai/openpilot #1490: Dashcam mode car unrecognized - 2019 jeep grand Cherokee trailhawk

cwarkenton, 2020-05-12:
OP worked great. Latest forced update to 7.3 gave dashcam mode car unrecognized.

VirtuallyChris, 2020-05-12:
What is your dongle ID?

cwarkenton, 2020-05-12:
Thanks! Shows N/A

cwarkenton, 2020-05-12:
Eon, Grey Panda, Giraffe, and FCA harness. All 'stock' from comma.ai

VirtuallyChris, 2020-05-12:
Dongle ID can be found in device Settings -> Device -> Dongle ID

cwarkenton, 2020-05-12:
04c3ecffabf06cdd

VirtuallyChris, 2020-05-13:
This appears to be fixed in 0.7.5 with this pull request:
https://github.com/commaai/openpilot/pull/1354

Just to be safe, I've added a few more of your car's messages into this pull request:
https://github.com/commaai/openpilot/pull/1496

cwarkenton, 2020-05-13:
Thanks for your help! 
I did pull and update add fingerprint through ssh. When the car started eon would give the proper 'keep eyes on road and be ready to take over' like it was good to go. When I get above around 8 mph it would give LKAS fault. 
Is 7.5 out somewhere I can flash?

VirtuallyChris, 2020-05-13:
It's out in release2-staging or wait until it comes out on release very soon

closed 2020-05-15 by pd0wm