commaai/openpilot #1495: Please connect to Internet message

ch1derco, 2020-05-13:
**Describe the bug**
A message "Please connect to Internet, An update Check is Required to Engage" comes after about 2 weeks after software installation. A factory reset and new install solves the problem for another week or 2, then message appears again. System is connected to the internet, but message will not go away. Problem exist since a couple of software versions..started in late 0.6 or early 0.7.x releases.

**How to reproduce or log data**
After about 2 weeks installation, message appears.

**Expected behavior**
No message

**Device/Version information (please complete the following information):**
 - Device: [EON Gold]
 - Version: [e.g. 0.7.4]
 - Car make/model [Chrysler Pacifica Hybrid
[image]
[image]
[image]

]

**Additional context**
Add any other context about the problem here.

pd0wm, 2020-05-13:
What is your dongle id? Can be found in device -> settings -> dongle id

ch1derco, 2020-05-13:
6fb4948a7ebe670e

“If you see something good, give a compliment. If you see something wrong, offer your help.” Nelson Mandela

pd0wm, 2020-05-13:
I see very few instances of the updater actually trying to run. Is your network blocking ping/ICMP packets? I've seen that with another user.

Before updating the updater tries to ping 8.8.8.8 to check if it's online:
https://github.com/commaai/openpilot/blob/master/selfdrive/updated.py#L338

pd0wm, 2020-05-22:
Closing due to inactivity.

closed 2020-05-22 by pd0wm