commaai/openpilot #1513: JoyStick seems won't work for me.

cdcfyjh, 2020-05-15:
1. Run boardd_old.py
2. Run carcontrol_debugger.py
3. Run keyboard_controller(I don't have a joystick.)(steer updates by k_left and k_right, gas and brake update by k_up and k_down. num_0, num_1, num_2, num_3 update pcm_cancel_cmd, nabled, audible_alert,(audible_alert and hud_alert) )

**Expected behavior**
I want to control steer. But the car state shows it only change a little when I press k_left and k_right.

Start: steeringAngle:  -0.09999995678663254
Pressed: steeringAngle:  -0.5583999752998352

The steeringAngle max is 480. So something must be wrong.

**Device/Version information (please complete the following information):**
 - Device: Panda
 - Dongle ID: Unknown
 - Version: master
 - Car make/model Toyota Camryh

CP:
carName = "toyota",
  carFingerprint = "TOYOTA CAMRY HYBRID 2018",
  enableGasInterceptor = false,
  enableCruise = true,
  enableCamera = true,
  enableDsu = false,
  enableApgs = false,
  minEnableSpeed = -1,
  minSteerSpeed = 0,
  safetyModel = toyota,
  safetyParam = 73,
  steerMaxBP = [0],
  steerMaxV = [1],
  gasMaxBP = [0],
  gasMaxV = [0.5],
  brakeMaxBP = [0],
  brakeMaxV = [1],
  mass = 1678.2128,
  wheelbase = 2.82448,
  centerToFront = 1.2427713,
  steerRatio = 13.7,
  steerRatioRear = 0,
  rotationalInertia = 3140.4302,
  tireStiffnessFront = 163310.56,
  tireStiffnessRear = 202840.58,
  longitudinalTuning = (
    kpBP = [0, 5, 35],
    kpV = [3.6, 2.4, 1.5],
    kiBP = [0, 35],
    kiV = [0.54, 0.36],
    deadzoneBP = [0, 9],
    deadzoneV = [0, 0.15] ),
  lateralTuning = (
    pid = (kpBP = [0], kpV = [0.6], kiBP = [0], kiV = [0.1], kf = 6e-05) ),
  steerLimitAlert = false,
  vEgoStopping = 0,
  directAccelControl = false,
  stoppingControl = false,
  startAccel = 0,
  steerRateCost = 1,
  steerControlType = torque,
  radarOffCan = false,
  steerActuatorDelay = 0.12,
  openpilotLongitudinalControl = false,
  carVin = "LVGBT74K7KG047683",
  isPandaBlack = true,
  dashcamOnly = false,
  safetyModelPassive = silent,
  transmissionType = unknown,
  carFw = [
    ( ecu = engine,
      fwVersion = "\x028966306R1100\x00\x00\x00\x00897CF3302002\x00\x00\x00\x00",
      address = 1792,
      subAddress = 0 ),
    ( ecu = esp,
      fwVersion = "F152633B80\x00\x00\x00\x00\x00\x00",
      address = 1968,
      subAddress = 0 ),
    ( ecu = dsu,
      fwVersion = "8821F0605200    ",
      address = 1937,
      subAddress = 0 ),
    ( ecu = eps,
      fwVersion = "8965B33541\x00\x00\x00\x00\x00\x00",
      address = 1953,
      subAddress = 0 ),
    ( ecu = fwdRadar,
      fwVersion = "8821F0605200    ",
      address = 1872,
      subAddress = 15 ),
    ( ecu = fwdCamera,
      fwVersion = "8646F0604100    ",
      address = 1872,
      subAddress = 109 ) ],
  radarTimeStep = 0.05,
  communityFeature = false,
  steerLimitTimer = 0.4,
  fingerprintSource = fw,
  networkLocation = fwdCamera

pd0wm, 2020-05-15:
Can you follow the latest instruction in the readme and report back (especially the part about the panda):

> Use the panda's OBD-II port to connect with your car and a usb cable to connect the panda to your pc. Also, connect a joystick to your pc.
> 
> joystickd.py runs a deamon that reads inputs from a joystick and publishes them over zmq. boardd sends the CAN messages from your pc to the panda. debug_controls is a mocked version of controlsd.py and uses input from a joystick to send controls to your car.
> 
> Make sure the conditions are met in the panda to allow controls (e.g. cruise control engaged). You can also make a modification to the panda code to always allow controls.
> 
> Usage:
> 
> python carcontrols/joystickd.py
> 
> # In another terminal:
> PARAMS_PATH=persist/params selfdrive/boardd/boardd
> 
> # In another terminal:
> python carcontrols/debug_controls.py

cdcfyjh, 2020-05-18:
I connect panda to car and pc correctly. Sees a blue and red LEDs twinkle. The boardd sends message get from zmq 'sendcan' to address 740. When I press the keyboard, the contain changes, so the zmq should be okay. 
What do you want to know about the panda? I can only see the LEDs.

pd0wm, 2020-05-22:
I don't see any changes in boardd about the change to the safety mode. Are you starting with the correct PARAMS_PATH env variable? (`PARAMS_PATH=persist/params selfdrive/boardd/boardd
`)

pd0wm, 2020-06-19:
Closing due to inactivity.

closed 2020-06-19 by pd0wm

javisaez95, 2022-01-18:
Hi!

I've been trying to use the joystick mode in a 2019 Toyota prius hybrid. For that purpose I'm using a black panda and a toyota car harness. I flashed the black panda firmware to be in debug mode to set controls_allowed to True always. I'm not able to engage openpilot in joystick mode unless I have the cruise mode on and working. The cruise mode only works when the car is above 25 mph, but I want to engage as well when the car is stopped. Is it possible?

Furthermore, when it is engaged (being above 25 mph) and the screen shows: "Joystick mode: Steer X% Gas X%" the car does not respond to the orders I sent to it, but the values of the Steer and gas percentage change on the screen. Why can it be happening?

fervh, 2024-04-19:
@javisaez95 Did you get the joystick to work with the black panda?