commaai/openpilot #1520: Add a few missing acryonyms to documentation

jjcf89, 2020-05-16:
The README defines quite a few acronyms but still seems to be missing some from the supported car lists

> openpilot is an open source driver assistance system. Currently, openpilot performs the functions of Adaptive Cruise Control (ACC), Automated Lane Centering (ALC), Forward Collision Warning (FCW) and Lane Departure Warning (LDW) for a growing variety of supported car makes, models and model years. In addition, while openpilot is engaged, a camera based Driver Monitoring (DM) feature alerts distracted and asleep drivers.

Undefined acronyms:
* SCC - Smart Cruise Control
* LKAS - Lane Keep Assist System?
* LDWS - Lane Departure Warning System?

VirtuallyChris, 2020-05-18:
Thanks for the comment jjcf89!

These terms will be better defined in the community wiki that will be launching later this week. :)

adeebshihadeh, 2020-05-25:
All three were added to the new wiki. See https://github.com/commaai/openpilot/wiki/General-Terms

closed 2020-05-25 by adeebshihadeh