commaai/openpilot #1579: Files in common/* violate PEP8.

dapperfu, 2020-05-28:
**Describe the bug**

Files in common/* are not currently being checked by the flake8 script.

There are files in it that violate pep8:

```
Pyflakes found errors in the code. Please fix and try again
common/tests/test_numpy_fast.py:3:1: F401 'timeit' imported but unused
common/window.py:30:24: F821 undefined name 'QUIT'
common/window.py:33:24: F821 undefined name 'KEYDOWN'
```

Additionally  files have 2 spaces to indent.
PEP8 clearly states 4 spaces. 

**Expected behavior**

Code should follow PEP8 guidelines.

adeebshihadeh, 2020-05-28:
Our flake8 check includes all the release files (`release/files_common`), which includes the files in common. openpilot's indentation level is an intentional style decision that happens to deviate from PEP8.

closed 2020-05-28 by adeebshihadeh

pd0wm, 2020-05-28:
I think it's good if we expand our coverage of the flake8 test to the whole repo, I'm doing that for pylint in this PR: https://github.com/commaai/openpilot/pull/1580/files.