commaai/openpilot #1610: No module named 'capnp'

kaishijeng, 2020-05-31:
When I run scons -j4 and got the following error:

    ModuleNotFoundError: No module named 'capnp'

How can I build capnp python module?

Thanks

robelkebede, 2020-05-31:
install capnp using: pip install pycapnp

kaishijeng, 2020-05-31:
@robelkebede 

Yes, it works after installing pycapnp.
However, I encountered another issue after finishing scons and run  "PASSIVE=0 NOBOARD=1 python ./manager.py" in selfdrive directory:

subprocess.CalledProcessError: Command 'cd /home/user/2TB/openpilot/panda/python/../board && : && make -f Makefile obj/panda.bin' returned non-zero exit status 2.
Exception ignored in: <function TextWindow.__del__ at 0x7f2485403050>
Traceback (most recent call last):
  File "/home/user/2TB/openpilot/common/text_window.py", line 40, in __del__
  File "/home/user/2TB/openpilot/common/text_window.py", line 29, in close
AttributeError: 'FakeTextWindow' object has no attribute 'text_proc'

Any idea how to fix it?

Thanks,

pd0wm, 2020-05-31:
You probably don't have `gcc-arm-none-eabi` installed. Try running `make` manually from `panda/board` to see what the error is. There seems to be a bug with the FakeTextWindow when it tries to show the error message. I'll look into that.

pd0wm, 2020-05-31:
TextWindow bug is fixed: https://github.com/commaai/openpilot/commit/0e7d50941551d27c8f12e618fac0c03a96109f07

kaishijeng, 2020-06-01:
@pd0wm 

Do I need to install gcc-arm-none-eabi? I just want to build to run carla simulator with openpilot?
After change text_window.py, the issue is resolved. However, another issue below happens:

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.
common/text_window.py: needs update
{"event": "dirty comma branch", "version": "0.7.6", "dirty": true, "origin": "https://github.com/commaai/openpilot.git", "branch": "origin/master", "dirty_files": ":100755 100755 50fdcb64404e06008f8ea83115e0a79e73d45da1 0000000000000000000000000000000000000000 M\tcommon/text_window.py\n", "commit": "b08fff33a0d8e05cb0941c69ecbc0bbff89783c6", "origin_commit": "b08fff33a0d8e05cb0941c69ecbc0bbff89783c6"}
sh: 1: content: not found
call: unrecognized service
getting pilotauth
dongle id is 425d2d0b832d2047
dirty is 1
preimporting selfdrive.thermald.thermald
Building panda firmware
Manager failed to start
Traceback (most recent call last):
  File "./manager.py", line 607, in <module>
    main()
  File "./manager.py", line 582, in main
    manager_prepare(spinner)
  File "./manager.py", line 525, in manager_prepare
    prepare_managed_process(p)
  File "./manager.py", line 321, in prepare_managed_process
    importlib.import_module(proc)
  File "/home/user/anaconda3/envs/dl/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/user/2TB/openpilot/selfdrive/thermald/thermald.py", line 22, in <module>
    FW_SIGNATURE = get_expected_signature()
  File "/home/user/2TB/openpilot/selfdrive/pandad.py", line 24, in get_expected_signature
    fw_fn = get_firmware_fn()
  File "/home/user/2TB/openpilot/selfdrive/pandad.py", line 18, in get_firmware_fn
    build_st(fn, clean=False)
  File "/home/user/2TB/openpilot/panda/python/__init__.py", line 34, in build_st
    _ = subprocess.check_output(cmd, stderr=subprocess.STDOUT, shell=True)
  File "/home/user/anaconda3/envs/dl/lib/python3.7/subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "/home/user/anaconda3/envs/dl/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'cd /home/user/2TB/openpilot/panda/python/../board && : && make -f Makefile obj/panda.bin' returned non-zero exit status 2.
Traceback (most recent call last):
  File "./manager.py", line 607, in <module>
    main()
  File "./manager.py", line 582, in main
    manager_prepare(spinner)
  File "./manager.py", line 525, in manager_prepare
    prepare_managed_process(p)
  File "./manager.py", line 321, in prepare_managed_process
    importlib.import_module(proc)
  File "/home/user/anaconda3/envs/dl/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/user/2TB/openpilot/selfdrive/thermald/thermald.py", line 22, in <module>
    FW_SIGNATURE = get_expected_signature()
  File "/home/user/2TB/openpilot/selfdrive/pandad.py", line 24, in get_expected_signature
    fw_fn = get_firmware_fn()
  File "/home/user/2TB/openpilot/selfdrive/pandad.py", line 18, in get_firmware_fn
    build_st(fn, clean=False)
  File "/home/user/2TB/openpilot/panda/python/__init__.py", line 34, in build_st
    _ = subprocess.check_output(cmd, stderr=subprocess.STDOUT, shell=True)
  File "/home/user/anaconda3/envs/dl/lib/python3.7/subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "/home/user/anaconda3/envs/dl/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'cd /home/user/2TB/openpilot/panda/python/../board && : && make -f Makefile obj/panda.bin' returned non-zero exit status 2.

Thanks,

robelkebede, 2020-06-01:
__init__ method in /openpilot/common/text_window.py will throw OSError exception. clone openpilot again and it will work

pd0wm, 2020-06-01:
@kaishijeng You need to install it, or comment out pandad in manager. As you can see from your error log it still fails running `'cd /home/user/2TB/openpilot/panda/python/../board && : && make -f Makefile obj/panda.bin`

kaishijeng, 2020-06-04:
It will be better to have an option in the build just for PC environment.
Thanks,