commaai/openpilot #1614: question about the function "model_eval_frame"

YJCITA, 2020-06-01:
When I read the code, finde the func "ModelDataRaw model_eval_frame", which calculate pos(camera trans_vs, rot_vs ). I veru interting how to calculate, but I get nothing from the code. So can you share some details or papaers about this?
Thx

pd0wm, 2020-06-01:
`model_eval_frame` runs the neural net on an image from the camera. Can you be more specific about your question?

YJCITA, 2020-06-02:
> `model_eval_frame` runs the neural net on an image from the camera. Can you be more specific about your question?

 use the neural net to calculate camera motion information(such as  trans_vs, rot_vs) directly?
I wonder is there use some algorithms llike: Visual Odometry, or Visual Inertial Odemotry?

pd0wm, 2020-06-02:
VIsual odometry is used in to generate the ground truth for the model, this not used online. The online model just takes in two frames and predicts the translation and rotation.

https://arxiv.org/abs/1812.05752v1 has some more info about our ground truthing stage + references to other papers.

YJCITA, 2020-06-03:
> model

thx!!
Another question: do you evaluate the model output translation and rotation precision in different road case? How aboult he result?

pd0wm, 2020-06-03:
Can you elaborate on that question? Not sure what you mean exactly.

YJCITA, 2020-06-15:
I think I have get what I need

closed 2020-06-15 by YJCITA