commaai/openpilot #1623: no alert for car stock acc turning off when onroad starts

VirtuallyChris, 2020-06-02:
**Describe the bug**
It's been 30 hours, the comma two is off. When I turn on the car, the device begins to boot - but I still want to use my ACC for the first few minutes of my drive. I activate cruise control and am cruising on the highway. The comma two goes onroad, and stock acc immediately disengages with no warning.

**Expected behavior**
It would be great if there was some sort of alert from the comma two that stock ACC has been disengaged upon boot.

adeebshihadeh, 2020-06-25:
Seems like this is because carState initializes with values that create noEntry events which will cancel the cruise mode.

adeebshihadeh, 2020-12-19:
Will be addressed with #2262

closed 2020-12-19 by adeebshihadeh