commaai/openpilot #1630: Enable disengagement analysis from qlogs

geohot, 2020-06-03:
This is still WIP.

Currently the qlogs don't contain "plan" or "pathPlan". I propose to add them both. I understand plan is triggered on radarState. Can we make it based on camera and move longitudinal stuff to pathPlan?

Can we increase the localizer frequency to 20hz to match the camera and plan?

Also, if we trust liveLocationKalman, we no longer need gpsLocationExternal, gpsLocation, or cameraOdometry. If we don't have gpsLocationExternal, does the filter fall back to gpsLocation?

pd0wm, 2020-06-03:
why do you want to merge plan and pathPlan? Even if they are sent at the same time I still think it makes sense to separate them.

plan has very little to do with the camera, and think it should be triggered on radard output. Even if radard would be driven by the camera it would always be a cycle late due to the radard processing time.

geohot, 2020-06-03:
Right now plan has little to do with the camera, but as we move to integrate long and lat control, it will come more and more from the camera. Right now what happens in radarless cars?

Also, what happened to the PR to remove l_poly and r_poly from MPC? I really liked that change if it didn't impact anything.

HaraldSchafer, 2020-06-03:
We cannot increase rate at the moment. Would almost double cpu, since most cpu is in the output message generation. We will try to reduce that, then maybe we can.

Yes I think we can replace those with livelocationKalman once I finish a large batch analysis. Only did 2k so far, want to do ~50k.

It does not fall back. That gps still has too many failure modes that I don't completely understand. Would be some effort to make sure it isn't a problem, don't think it is worth the time and code complexity.

geohot, 2020-06-03:
Added in path and pathPlan to qlogs at 10hz. I figure that's the rate which we'll do all the disengagement analysis.

Spoke with Willem about 20hz localizer, will happen once CPU is better.

Okay on gpsLocation, really committed to the future then!

geohot, 2020-06-03:
Added validLen to PathData. Ideally we should start passing this downstream to MPC, I guess MPCs dependence on speed is a decent hack for this, but going forward we should use the actual value.

closed 2021-10-05 by HaraldSchafer