commaai/openpilot #1641: Intermittent temp steering faults for Toyota

jyoung8607, 2020-06-05:
Some users are seeing intermittent temp steering faults on sharp turns for Toyota. A solution in openpilot was attempted in #1256 but not yet accepted.

It's been reported that driving in all-output safety mode makes the problem go away. This is not yet fully verified, and obviously that's not a good daily-drive configuration, but it's of diagnostic interest. It tells us the Panda safety mode may be discarding messages for torque level or ramp-rate violations, and the temp steering faults from the steering rack could be a consequence of those missing messages.

If this can be verified, it means we need to find and fix two issues:

1) There's airspace between what OP can send and what Panda safety is willing to pass
2) There's airspace between what OP can send and Panda regression testing is exercising

If you're experiencing this issue, please add a comment with a drive segment log, and I will take a look early next week.

@wocsor -  fbd011384db5e669|2020-06-04--15-46-08--11
@ShaneSmiskol - ?

VirtuallyChris, 2020-06-05:
I've had a support request from the other day - this guy apparently gets steer unavalible all the time, as well as cruise faults which seems to be another issue (even though his DSU is plugged in)

cf0938edab43749f

husker91kyle, 2020-06-05:
This has been happening to me... How can I post my route for you guys to see?
My dongle ID is 013e60cac3cd93af

VirtuallyChris, 2020-06-05:
That's all you need to send, thanks! @husker91kyle

pd0wm, 2020-06-05:
@husker91kyle Seems like you were running shane's fork based on 0.7.5 So a different problem than #1654. Can you verify the problem exists when you run release2 instead of a fork?

pd0wm, 2020-06-05:
Closed this issue. Think there were three different issues, of which one is solved now. Let's create new issues for those. I also recommend strongly against driving in all output mode. If there are issues with the safety mode it can be quickly verified offline that messages are blocked.

To summarize there were three issues
1) Issue of dropped packets on 0.7.6 devel which @wocsor and some other users on discord were  seeing, should be fixed now
2) @husker91kyle issue running 0.7.5. I suspect a wiring issue, but would need to look at more data.
3) People strongly and quickly overriding on Toyota while in a sharp turn causing faults.

closed 2020-06-05 by pd0wm