commaai/openpilot #1669: Poor focus in snapshots

csouers, 2020-06-09:
**Example snapshots**
[image]
[image]

**Still from driving video**
[image]

**Describe the bug**
Focus appears to be off significantly in rear camera snapshots

**How to reproduce or log data**
Capture snapshot via athena.

**Expected behavior**
Snapshot with good focus.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 95e637a323d6e9fc
 - Version: master-ci branch based off of 9d8ff2887c280388eb0abc589d5f562a6dfac0ad
 - Car make/model: Civic Hatchback 2017

**Additional context**
Focus during normal driving footage is fine and looks great. Only the snapshots appear to be affected. I query snapshots constantly throughout the day and it always exhibits the same behavior. I thought maybe it was compression at play, but I'm starting to thing that it's not the issue.

pd0wm, 2020-06-09:
It probably doesn't wait long enough for camerad to be completely focussed. Does it help if you increase the 3 second timeout to something large like 10 seconds?https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/snapshot/snapshot.py#L41

Camerad now outputs a sharpness score. We should probably change the snapshot code to subscribe to that and detect when the image is properly focused before taking the snapshot (obviously with some max time).

adeeb10abbas, 2021-01-09:
Does the 3-second timeout window solution work? If yes, can this issue be closed?

pd0wm, 2021-01-11:
No, the three seconds is not enough.

If you want to fix it you'd need to update the code here https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/snapshot/snapshot.py#L37 to not exit the while loop until the sharpnessScore of the rear camera is above some threshold. Of course it also needs a max timeout of something like 10 seconds in case it never focuses when the device is on a desk.

adeeb10abbas, 2021-01-13:
I get how it would work. I am not able to figure out where is the "sharpness score" that you mentioned earlier. All I can find is this here[https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/test/test_camerad.py#L53] 
```
  def _is_really_sharp(self, i, threshold=800, roi_max=np.array([8,6]), roi_xxyy=np.array([1,6,2,3])):
    i = self._numpy_bgr2gray(i)
    x_pitch = i.shape[1] // roi_max[0]
    y_pitch = i.shape[0] // roi_max[1]
    lap = self._numpy_lap(i)
    lap_map = np.zeros((roi_max[1], roi_max[0]))
    for r in range(lap_map.shape[0]):
      for c in range(lap_map.shape[1]):
        selected_lap = lap[r*y_pitch:(r+1)*y_pitch, c*x_pitch:(c+1)*x_pitch]
        lap_map[r][c] = 5*selected_lap.var() + selected_lap.max()
    print(lap_map[roi_xxyy[2]:roi_xxyy[3]+1,roi_xxyy[0]:roi_xxyy[1]+1])
    if (lap_map[roi_xxyy[2]:roi_xxyy[3]+1,roi_xxyy[0]:roi_xxyy[1]+1] > threshold).sum() > \
          (roi_xxyy[1]+1-roi_xxyy[0]) * (roi_xxyy[3]+1-roi_xxyy[2]) * 0.9:
      return True
    else:
      return False
```
In this case, it seems to be calculating the score and then comparing it. Let me know. Thanks!

advcu987, 2021-04-04:
I'm also taking a look at it, hope you don't mind. 
In cereal/log.capnp you have a `struct FrameData`, that contains the `sharpnessScore` field. I guess this is what is needed.

closed 2021-05-27 by ShaneSmiskol