commaai/openpilot #1670: HKG sensor data / permanent can error check connections

AskAlice, 2020-06-09:
**Describe the bug**
A clear and concise description of what the bug is.
So I've personally experienced a pair of permanent can error alerts as well as device sensor alerts with 0.8. Presumably this began around 0.7.6. On 0.7.5 it was working great up until a couple days before release, then I started getting 'can error check connections' for that.

None of these seem to provide me with any useful tracing in tmux.

I do not have long control, so generally my build requires a couple tweaks before it can engage, but @KyleGreenlaw reported to me that with his vehicle that does have SCC, he was unable to get the latest and greatest running on HKG as well.

[image]
[image]

eb1f2dd16c2313e6|2020-06-09--02-56-34
https://my.comma.ai/eb1f2dd16c2313e6/1591692995041/1591693110984

**How to reproduce or log data**
Steps to reproduce the behavior, or a explorer/cabana link to the exact drive and timestamp of when the bug occurred.
get in a 2019 forte or a 2018 elantra and run anything newer than 0.7.4
**Expected behavior**
A clear and concise description of what you expected to happen.
`alerts = []`
**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold] comma two, and eon (both tested)
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID] eb1f2dd16c2313e6
 - Version: [e.g. 0.6.4] 1a2411be0fbbc0db91914ff7dfc29e156471d7ab
 - Car make/model [e.g. Toyota Prius 2016] 2019 kia forte

**Additional context**
Add any other context about the problem here.

pd0wm, 2020-06-09:
It's because your car doesn't have SCC. Currently openpilot does not perform longitudinal control and expects the SCC messages for part of the carState. Specifically the `SCC11` and `SCC12` message. When I merged the hyundai port into openpilot I added checks that requires all the messages that are used in carState to be present (https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/carstate.py#L199). The hkg community fork works for you because those checks are not present and uses some default values for those fields.

So it seems like until we add KHG longitudinal support, your car is not supported by openpilot.

AskAlice, 2020-06-09:
got it, thanks

closed 2020-06-09 by AskAlice