commaai/openpilot #1672: Errors when running with Carla in a docker container

FHof, 2020-06-10:
**Describe the bug**

I'm trying to run the openpilot simulation but several errors appear in the manager when I start the carla bridge.

Here's the full manager.py output: https://pastebin.com/raw/T0Rq0hAZ
When I start the bridge, the UI window disappears.
Three of the threads are marked in red at first: logmessaged, ui, modeld
Then four of them are red: logmessaged, ui, modeld, proclogd

Here are some of the error messages:
```
CalibrationParams file found but error encountered
Traceback (most recent call last):
  File "/openpilot/selfdrive/locationd/calibrationd.py", line 71, in __init__
    self.vp = vp_from_rpy(calibration_params["calib_radians"])
KeyError: 'calib_radians'
```

```
chrt: failed to set pid 452's policy: Operation not permitted
```

```
./_modeld: error while loading shared libraries: libSNPE.so: cannot open shared object file: No such file or directory
```
(That comment about the libSNPE error may be related: https://github.com/commaai/openpilot/issues/1364#issuecomment-639340702)

**How to reproduce or log data**

Here is the Dockerfile for the container: https://gist.github.com/FHof/2c06f9b353f2caa37a448adcf0d70607
I created the docker image with a recent openpilot version: commit c6bd1874503b26e8443ecb28afd6819b44684e78

**Expected behavior**
I assume all threads should be marked green and the window should not close.

pd0wm, 2020-06-19:
It should use tensorflow to run the models. You should probably figure out why it decides to use snpe. If it's not runnning on `aarch64` it shouldn't use snpe: https://github.com/commaai/openpilot/blob/master/selfdrive/modeld/SConscript#L22

HybridDog, 2020-06-20:
I don't know the scons language; if I understand the code correctly, it only removes the SNPE model and library if the architecture is `Darwin`.

FHof, 2020-06-20:
I don't know the scons language; if I understand the code correctly, it only removes the SNPE model and library if the architecture is `Darwin`.

FHof, 2020-06-30:
After I have applied [those code changes](https://gist.github.com/FHof/1d3db42a5ff35a53bec8761b5e1e77e9), the calibrationd and SNPE error messages are gone, but it still does not work: https://pastebin.com/raw/1t69VRFt
logmessaged, ui and modeld are red.
I assume the calibrationd crash has been introduced at https://github.com/commaai/openpilot/pull/1543/files#diff-d93b28a3ff6b86d5e0cf71058c5d0703R71.

I have [changed some OpenCL messages to LOGE](https://pastebin.com/raw/XL0LDTp2) and noticed that device_type is set to CL_DEVICE_TYPE_CPU so that it does not load that OpenCL platform. When I set it to CL_DEVICE_TYPE_ALL, I get this output: https://pastebin.com/raw/HyP2700C
logmessaged, ui, calibrationd and modeld were red.
The [clinfo output](https://pastebin.com/raw/s3qGMcdf) shows that no CPU platforms are available.

FHof, 2020-07-02:
To get rid of the OpenCL failure, I have installed pocl-opencl-icd, which provides OpenCL on CPU, and removed the nvidia OpenCL platform. It still does not work; modeld is the only red one now and there is a visionstream error:
```
selfdrive/modeld/modeld.cc: visionstream connect failed
Assertion failed: ok (src/mailbox.cpp:99)
```
Here is full output: https://pastebin.com/raw/vv4WB68N

ed2k, 2020-07-12:
> Edit: I have tried the latest openpilot; after waiting a bit, errors appear with CL_DEVICE_TYPE_ALL:

you have GPU, so use `CL_DEVICE_TYPE_GPU`
like https://github.com/ed2k/openpilot/blob/test_keras/selfdrive/modeld/modeld.cc#L123

ed2k, 2020-07-12:
> Assertion failed: ok (src/mailbox.cpp:99)

#1655

closed 2020-09-10 by pd0wm

sohamm17, 2020-10-04:
I am trying to run the docker image on Intel GPU. I have Carla running on my Ubuntu 16.04.
I am trying to run OpenPilot with the `start_openpilot_docker.sh` but I get the following output:
```
selfdrive/modeld/modeld.cc: visionstream connect failed
WARNING:tensorflow:No training configuration found in the save file, so the model was *not* compiled. Compile it manually.
<tensorflow.python.keras.engine.training.Model object at 0x7f4b0e0c52b0>
ready to run keras model [TensorShape([1, 12, 128, 256]), TensorShape([1, 8]), TensorShape([1, 2]), TensorShape([1, 512])]
selfdrive/modeld/modeld.cc: visionstream connect failed
selfdrive/modeld/modeld.cc: visionstream connect failed
```
Any pointer?