commaai/openpilot #1684: Getting "White Panda Is No Longer Supported" when running with Carla

johschmitz, 2020-06-12:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**How to reproduce or log data**
Run openpilot with Carla as described here:
https://github.com/commaai/openpilot/tree/master/tools/sim

**Expected behavior**
It should drive in Carla.

**Device/Version information (please complete the following information):**
 - Device: PC
 - Version: 97b1ee25a7cff52216f5a1e8c5cb2b4b5847eba2

**Additional context**
-

pd0wm, 2020-06-12:
https://github.com/commaai/openpilot/commit/6810333075694c9c72d3c689f05462f6e6565e88 should fix it.

johschmitz, 2020-06-15:
I have the latest master branch including that fix but it doesn't seem to help.

johschmitz, 2020-06-15:
Okay I just went through the whole process from scratch, including deleting the openpilot folder and the pipenv cache, and now I don't get the error anymore. Not sure why.
[image]

What doesn't work though is speeding up. Documentation says
```
1: Cruise up 5 mph
2: Cruise down 5 mph
3: Cruise cancel
q: Exit all
```
But nothing happens.

johschmitz, 2020-06-16:
Okay, apparently the keyboard polling thread is running as a part of the Carla bridge (bridge.py)
So the answer is to have the focus on the bridge.py terminal when pressing 1, 2, 3, q.

closed 2020-06-16 by johschmitz