commaai/openpilot #1687: Connection issues on startup at high temps (75 degrees C)

Hubblesphere, 2020-06-12:
**Describe the bug**
Temporary dash error, switching between offroad/onroad while vehicle is running (Mid-day hot temps, C2 around 75 degrees C)

**How to reproduce or log data**
Start of this drive: f91c796ac0bf5eb6|2020-06-12--12-12-48--0

**Expected behavior**
Start overheated, maintain connection and engage once cooled down. 

**Device/Version information (please complete the following information):**
 - Device: [Comma Two]
 - Dongle ID: [f91c796ac0bf5eb6]
 - Version: [0.8 master-ci]
 - Car make/model [2019 Toyota Corolla Hatchback]

pd0wm, 2020-06-12:
Can you make sure the drives are uploaded?

pd0wm, 2020-06-12:
Looking at the qlogs the tempature of the device was about 80 C. When openpilot went onroad CPU temperature quickly spiked to over 107 C. We then go offroad to protect the CPU from overheating too much. The cpu temperature then probably dropped, we went onroad again and the cycle repeated. The solution would be to add hysteris to the `danger` thermal state so we don't go onroad when we risk overheating too much when we would do that.

Hubblesphere, 2020-06-12:
I'll have drives uploaded later this afternoon. I sometimes don't have a good connection to upload at work. Sounds easy enough to fix!

pd0wm, 2020-06-12:
https://github.com/commaai/openpilot/pull/1688

pd0wm, 2020-06-12:
Merged a fix. Please let us know if you like the new behavior of cooling down before going onroad.

closed 2020-06-12 by pd0wm

Hubblesphere, 2020-06-22:
This issue is still occurring due to temp spike on startup: f91c796ac0bf5eb6|2020-06-22--13-25-34--0

More stable as it only happens once, but throws a dash error until it cools and comes back on road.