commaai/openpilot #1696: HKG Resume Bug

TK211X, 2020-06-14:
[image]

When using stock HKG SCC there is a 1 in 20 chance the ECU will fault and cancel SCC when OP sends resume command. It has come to my attention that this usually happens after OP sends a resume command multiple times but the lead vehicle creeps within 3.7 meters and the SCC logic does not resume because of this. I would try modifying the code so that it remains as is, but does not attempt to resume unless the lead vehicle distance is reported as more than 3.7 meters and increasing. This may help remedy the issue although it may not be the root cause. Sending resume at and below 3.7 Meters is pointless as SCC will not resume anyway.

Example logic:

ACC_ObjDist

Stationary at X.X climbs to 3.7 <--- DO NOT RESUME

Stationary at X.X climbs to 3.8 <--- SEND RESUME PAST 3.7 (3.8)

Stationary at 4.8 <--- DO NOT RESUME

Stationary at 4.8 then climbs <--- SEND RESUME ON INCREASE (Greater than 3.7)

xx979xx, 2020-06-17:
I am not sure that I get everything but this change to [selfdrive/car/hyundai/carcontroller.py#L86](https://github.com/commaai/openpilot/blob/dbc609b240e598b06aec82c25d761d18d9c022ee/selfdrive/car/hyundai/carcontroller.py#L86) should help:
```
      elif self.last_lead_distance < CS.lead_distance > 4.8 and (frame - self.last_resume_frame) > 5:
```

TK211X, 2020-06-21:
Wouldn’t it be:

S.lead_distance > 3.7

@xx979xx

xps-genesis, 2020-06-25:
a res hit is valid for 2 seconds, if the lead starts to move, hit res(spam), then we hope the lead starts to move > 3.7m with in 2 sec. 

but right now we keep spamming as the lead distance keep changing.

xx979xx, 2020-06-26:
@TK211X what about this:

> Stationary at 4.8 <--- DO NOT RESUME

I don't get it

TK211X, 2020-06-26:
@xx979xx 

The SCC has stopped and the lead vehicle is at 4.8 in the example and not moving forward so we would not hit resume.

TK211X, 2020-06-26:
The resume bug appears to be due to the parity bit.

xps-genesis, 2020-06-26:
[image]

adeebshihadeh, 2020-07-07:
@xps-genesis is this not an issue on the community fork?

xps-genesis, 2020-07-07:
i haven't seen this issue on the community branch. that said, this is a very random issue. so no idea if that change shown in the screenshot above really helps.

PR #1852 (2020-07-10): Simplify hyundai resume logic
resolves #1696
closed 2020-07-10 by adeebshihadeh