commaai/openpilot #1705: how to log debug message on EON

YJCITA, 2020-06-15:
I add some my code in openpilot, but it not run correctly, so I want log some debug message on EON , but I cannot find the way?
I find there is a "qlog.bz2" and "rlog.bz2", but I donot know how the log and how to paeser.

adhintz, 2020-06-15:
One option is to just log to a file and `tail -f` it or look at it later. Here's an example: https://github.com/adhintz/openpilot/commit/4468669eca8c009b5d3fa5bdee28448fddb0aadf

YJCITA, 2020-06-18:
> One option is to log to a file and `tail -f` it or look at it later. Here's an example: [adhintz@4468669](https://github.com/adhintz/openpilot/commit/4468669eca8c009b5d3fa5bdee28448fddb0aadf)

it works,  thx

closed 2020-06-18 by YJCITA